FreeCAD/FreeCAD #19655: OndselSolver: hard crash on any new object

benj5378, 2025-02-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Reproduce
1. Open AssemblyExample.FCStd from Start page
2. Add any new object (new Body)
3. Observe the crash

While bugs in OndselSolver qualifies as third-party bugs, it must be considered if it's desirable that `SIGABRT` propagates up till FreeCAD crashes.

Issue only occurs in my own build.

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40236 (Git)
Build type: Debug
Branch: baseFwd
Hash: f12f521920928f0b4e9b3770dd3bb71bf7a55bec
Python 3.12.3, Qt 6.6.2, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical/physical DPI: 96/80.9812
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * fasteners 0.5.29
  * Help 1.0.3
```

### Subproject(s) affected?

None

### Anything else?

<details>
<summary>Backtrace</summary>

```cpp
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=<optimized out>, signo=signo@entry=6) at ./nptl/pthread_kill.c:89
#3  0x00007fffefe4527e in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#4  0x00007fffefe288ff in __GI_abort () at ./stdlib/abort.c:79
#5  0x00007fffefe2881b in __assert_fail_base
    (fmt=0x7fffeffd01e8 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", assertion=assertion@entry=0x7fff9ca502b4 "false", file=file@entry=0x7fff9ca50260 "/home/bensay/Development/FreeCAD/src/3rdParty/OndselSolver/OndselSolver/LimitIJ.cpp", line=line@entry=112, function=function@entry=0x7fff9ca50558 "virtual void MbD::LimitIJ::fillqsudot(MbD::FColDsptr)") at ./assert/assert.c:96
#6  0x00007fffefe3b517 in __assert_fail
    (assertion=0x7fff9ca502b4 "false", file=0x7fff9ca50260 "/home/bensay/Development/FreeCAD/src/3rdParty/OndselSolver/OndselSolver/LimitIJ.cpp", line=112, function=0x7fff9ca50558 "virtual void MbD::LimitIJ::fillqsudot(MbD::FColDsptr)") at ./assert/assert.c:105
#7  0x00007fff9c98433c in MbD::LimitIJ::fillqsudot
    (this=0x55555aec6980, col=std::shared_ptr<MbD::FullColumn<double>> (use count 2, weak count 0) = {...})
    at /home/bensay/Development/FreeCAD/src/3rdParty/OndselSolver/OndselSolver/LimitIJ.cpp:112
#8  0x00007fff9ca3089e in operator()
    (__closure=0x7fffffffa010, item=std::shared_ptr<MbD::Item> (use count 3, weak count 0) = {...})
    at /home/bensay/Development/FreeCAD/src/3rdParty/OndselSolver/OndselSolver/VelICSolver.cpp:65
#9  0x00007fff9ca32d44 in std::__invoke_impl<void, MbD::VelICSolver::runBasic()::<lambda(std::shared_ptr<MbD::Item>)>&, std::shared_ptr<MbD::Item> >(std::__invoke_other, struct {...} &) (__f=...)
    at /usr/include/c++/13/bits/invoke.h:61
#10 0x00007fff9ca3225f in std::__invoke_r<void, MbD::VelICSolver::runBasic()::<lambda(std::shared_ptr<MbD::Item>)>&, std::shared_ptr<MbD::Item> >(struct {...} &) (__fn=...) at /usr/include/c++/13/bits/invoke.h:111
#11 0x00007fff9ca31926 in std::_Function_handler<void(std::shared_ptr<MbD::Item>), MbD::VelICSolver::runBasic()::<lambda(std::shared_ptr<MbD::Item>)> >::_M_invoke(const std::_Any_data &, std::shared_ptr<MbD::Item> &&)
    (__functor=..., __args#0=...) at /usr/include/c++/13/bits/std_function.h:290
#12 0x00007fff9ca12943 in std::function<void (std::shared_ptr<MbD::Item>)>::operator()(std::shared_ptr<MbD::Item>) const (this=0x7fffffffa010, __args#0=std::shared_ptr<MbD::Item> (empty) = {...})
    at /usr/include/c++/13/bits/std_function.h:591
#13 0x00007fff9ca115a0 in std::for_each<__gnu_cxx::__normal_iterator<std::shared_ptr<MbD::LimitIJ>*, std::vector<std::shared_ptr<MbD::LimitIJ>, std::allocator<std::shared_ptr<MbD::LimitIJ> > > >, std::function<void (std::shared_ptr<MbD::Item>)> >(__gnu_cxx::__normal_iterator<std::shared_ptr<MbD::LimitIJ>*, std::vector<std::shared_ptr<MbD::LimitIJ>, std::allocator<std::shared_ptr<MbD::LimitIJ> > > >, __gnu_cxx::__normal_iterator<std::shared_ptr<MbD::LimitIJ>*, std::vector<std::shared_ptr<MbD::LimitIJ>, std::allocator<std::shared_ptr<MbD::LimitIJ> > > >, std::function<void (std::shared_ptr<MbD::Item>)>)
    (__first=std::shared_ptr<MbD::LimitIJ> (use count 3, weak count 0) = {...}, __last=std::shared_ptr<MbD::LimitIJ> (empty) = {...}, __f=...) at /usr/include/c++/13/bits/stl_algo.h:3833
#14 0x00007fff9ca0bfdd in MbD::System::partsJointsMotionsLimitsDo(std::function<void (std::shared_ptr<MbD::Item>)> const&) (this=0x55555ae4e7a0, f=...)
    at /home/bensay/Development/FreeCAD/src/3rdParty/OndselSolver/OndselSolver/System.cpp:163
[... 60 lines cut ...]
    (this=0x555555639df0, sCmd=0x55555ad2df50 "App.ActiveDocument.recompute()")
    at /home/bensay/Development/FreeCAD/src/Base/Interpreter.cpp:249
#40 0x00007ffff592efbd in Gui::Command::_runCommand
    (file=0x7ffff625a670 "/home/bensay/Development/FreeCAD/src/Gui/Command.cpp", line=851, eType=Gui::Command::App, sCmd=0x55555ad2df50 "App.ActiveDocument.recompute()")
    at /home/bensay/Development/FreeCAD/src/Gui/Command.cpp:725
#41 0x00007ffff592eb79 in Gui::Command::_doCommand
    (file=0x7ffff625a670 "/home/bensay/Development/FreeCAD/src/Gui/Command.cpp", line=851, eType=Gui::Command::App, sCmd=0x7ffff625a788 "App.ActiveDocument.recompute()")
    at /home/bensay/Development/FreeCAD/src/Gui/Command.cpp:677
#42 0x00007ffff5930036 in Gui::Command::updateActive ()
    at /home/bensay/Development/FreeCAD/src/Gui/Command.cpp:851
#43 0x00007fffa8582c5c in CmdPartDesignBody::activated (this=0x555555f4e3e0, iMsg=0)
    at /home/bensay/Development/FreeCAD/src/Mod/PartDesign/Gui/CommandBody.cpp:296
#44 0x00007ffff592d745 in Gui::Command::_invoke (this=0x555555f4e3e0, id=0, disablelog=true)
    at /home/bensay/Development/FreeCAD/src/Gui/Command.cpp:444
#45 0x00007ffff592d457 in Gui::Command::invoke (this=0x555555f4e3e0, i=0, trigger=Gui::Command::TriggerAction)
--Type <RET> for more, q to quit, c to continue without paging--
    at /home/bensay/Development/FreeCAD/src/Gui/Command.cpp:408
#46 0x00007ffff59099e2 in Gui::Action::onActivated (this=0x5555579ac8e0)

```
</details>

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-16:
No issue here. @PaddleStroke FYI
```
OS: macOS 13.7.3
Architecture: x86_64
Version: 1.1.0dev.40253 (Git) Conda
Build type: Release
Branch: main
Hash: e16c462916a050909ad107e4f1ac2e0e5990389d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell /.0.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 72/72
Installed mods: 
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * CfdOF 1.29.8
  * Defeaturing 1.2.3
  * Manipulator 1.5.7
  * fasteners 0.5.37
  * CurvedShapes 1.0.13
  * sheetmetal 0.7.11
  * Curves 0.6.58

```

oursland, 2025-05-30:
@benj5378 can you still reproduce this issue?

benj5378, 2025-06-11:
> [@benj5378](https://github.com/benj5378) can you still reproduce this issue?

No

closed 2025-06-11 by benj5378