FreeCAD/FreeCAD #19657: TechDraw: Insert View not doing anything, no view inserted

nv-steve, 2025-02-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When attempting to insert a body as a View for a Page, nothing happens. No error or warning, simply nothing.
I started with a forum post but it now seems to be a bug.
https://forum.freecad.org/viewtopic.php?p=811589

**Reproduction Steps:**
1. Open the attached file
2. Select any Body from the tree or 3D view
3. Change to Tech Draw workbench
4. Click Insert View command
5. Choose "ANSI B Size Test" as the target Page
6. Observe that nothing happens

**Expected result:**
The Body should be inserted as a new View.

[bookcase - RnD on corner vert partition - take 3 - blind dado rnd.FCStd.tar.gz](https://github.com/user-attachments/files/18815941/bookcase.-.RnD.on.corner.vert.partition.-.take.3.-.blind.dado.rnd.FCStd.tar.gz)

### Full version info

```shell
OS: Linux Mint 22 (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.40077 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: a85756b1c94c30252f99c7bc497beed5c8dcbf25
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96.0135/108.888
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

WandererFan, 2025-02-16:
I can not repeat this on demand.  I have seen it happen following the above steps - maybe once in every 10 attempts, maybe not that frequently.

@nv-steve from the forum thread, it appears that you can recreate this on demand?

steveklett, 2025-02-16:
@WandererFan Yes, I can repeat on demand.

steveklett, 2025-02-16:
While trying to create the video I think I realized what's happening. It's not a bug, it's me and the way my brain is registering the Page Choose UI dialog. 

When the Page Choose is displayed like the below image my brain was registering the "ANSI B Size Test (Page003)" as the selected Page.
[image]

However, it's not really selected. If you click on the list item it becomes dark blue:
[image]

and this is when the Insert will succeed. This was my mistake, as I said, when I see the light blue selection I interpreted that as already selected.

I think there is an opportunity for improvement with the Page Choose.

1. It seems this list control needs to have 1 of the items with the light blue selection. I don't know what that indicates as it's not really selected, so what is it? What is light blue background mean? If possible, all list items could look the same to make it obvious there isn't a selection.
2. I don't think a user would ever expect to hit OK from the Page Chooser and have nothing happen. A Page selection should be mandatory, the OK button disabled until a Page is selected. The user can always Cancel out of the dialog if they don't want to proceed.

@WandererFan I'm sorry for wasting your time with this issue and I'm a bit embarrassed I didn't realize what I was doing earlier.

Should this issue be converted to a feature request? How do we proceed?

WandererFan, 2025-02-17:
Still points out a missing error response if no page is selected, so not a waste of time.  I've updated the issue title.

I don't know exactly where the light blue comes from yet.  It might just be a cursor for the current line, but it certainly looks like it is selected.

benj5378, 2025-02-17:
@WandererFan 

After QListWidget setup, set selected item to first, such that there is no confusion on startup.
Disable OK button if none selected - in case someone should deselect all

closed 2025-02-23 by WandererFan