FreeCAD/FreeCAD #19658: Part Design: Additive Lost is twisted 180 degrees

nv-steve, 2025-02-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When creating an Additive Loft in Part Design WB the result is twisted 180 degrees. I have tried various methods to generate the source face or sketches used in the Loft, but all result in a twisted result.

The history is documented in this forum post:
https://forum.freecad.org/viewtopic.php?p=811519#p811519

**Reproduction Steps:**
1. Open file Twisted Loft Issue example.FCStd
2. Ensure Body012 is Active
3. Select the left face of Binder023
4. Invoke the Additive Loft tool in the Part Design WB
5. Click Add Section
6. Select the Right face of Binder023 and the loft should generate that looks like this:

[image]

I've tried creating new sketches to essentially trace the Binder faces and still the loft will be twisted.

**Expected Result:**
The loft should be generated as a flat, untwisted solid

[Twisted Loft Issue example.FCStd.tar.gz](https://github.com/user-attachments/files/18815965/Twisted.Loft.Issue.example.FCStd.tar.gz)

### Full version info

```shell
OS: Linux Mint 22 (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.40077 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: a85756b1c94c30252f99c7bc497beed5c8dcbf25
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96.0135/108.888
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-17:
Can you please create a simpler file? It appears one end is rotated 180 degrees

closed 2025-02-17 by maxwxyz

steveklett, 2025-02-17:
@maxwxyz I've been trying to reproduce your workaround of rotating the individual Binder. I'm assuming you used Transformation on the Binder, but I'm not able to rotate it "in place". Do you mind providing just a bit more detail on what you did to rotate the binder without it's position changing?

Also, while this appears to be a valid workaround, why is it necessary? I think if I draw two new sketches and the loft is still twisted, that is unexpected behavior.

I'm also doubting that the issue is due to a rotated Body or Clone, I'll work on a more simple example that reproduces the issue, if possible.

steveklett, 2025-02-17:
@maxwxyz I executed my process again to make sure I wasn't rotating anything that could result in the twisted loft. I produced a video that shows my steps (I can't make audio recording work on my laptop so it is a silent vide)
https://youtu.be/OoiG9XZd0Oo

I also distilled down my FC project file to the bare essentials and have attached it.
I don't think I've done anything in my models that should result in a twisted loft. Because this issue is closed I don't know if you're seeing my comments, I hope you are.  :)

[bookcase A - carcasses - flipped loft simple example - 2.FCStd.tar.gz](https://github.com/user-attachments/files/18830273/bookcase.A.-.carcasses.-.flipped.loft.simple.example.-.2.FCStd.tar.gz)

maxwxyz, 2025-02-18:
The issue is basically this: https://github.com/FreeCAD/FreeCAD/issues/12436

maxwxyz, 2025-02-18:
1. No
2. I have used the placement tool

Currently it is not working to specify the order of a loft. You can make it work like this:
Create a Sketch attached to the binders (each) and redraw (use external geometry tool for references) the profile. You can then loft it without a twist. Your first point of the first sketch will connect to the first point of the second sketch.
For further help use the forum or Discord.