FreeCAD/FreeCAD #19659: Core: Sync placement does not work when dropping in a Std Group nested in a Std Part

Roy-043, 2025-02-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

1. Make sure [SyncPlacement](https://wiki.freecad.org/Std_TreeSyncPlacement) is on.
2. Open the attached [file](https://github.com/user-attachments/files/18815955/sync_placement_test.FCStd.zip) (remove the .zip extension).
3. Drag the smaller circle out of the Group and drop it on the document node.
4. Result: Placement is synced. OK.
5. Drag the larger circle into the Group.
6. Result. Placement is not synced. Even though the object is added to both the Group property of the Group and of the Part.
7. Reopen the file.
8. Drag the larger circle into the Part.
9. Result: Placement is synced. The circle can now be dragged into the Group to achieve the correct final result.

### Full version info

```shell
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.1.0dev.40176 (Git) Conda
Build type: Release
Branch: main
Hash: 2745f436026b6f1fb84dbaaf89ea22b5fb4c2295
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell /.0.0, OCC 7.8.1
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/86.4394
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-17:
Was this working in 1.0, sounds like a regression

Roy-043, 2025-02-17:
I have checked V0.19, V1.0 and V1.1. In all these versions the issue occurs. So it seems to be an older bug.

Roy-043, 2025-11-04:
Still an issue:
```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.20251031 (Git shallow)
Build date: 2025/10/31 03:23:50
Build type: Release
Branch: main
Hash: c270a8ac666dff22a92baf2a3a4f5200197bb678
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```