FreeCAD/FreeCAD #19662: Unit Systems Incomplete Labeling

brendanm250, 2025-02-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When starting FreeCAD for the first time, the unit system options presented to the user have strange and presumably incomplete descriptions. Example below but several appear to be incomplete.

- Standard (mm, kg, s, deg)
- MKS (m, kg, s, deg)
- US Customary (in, lb)
  - Ex. of missing time and angle units

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Anything else?

I just downloaded for the first time so I don't know if this uniquely shows up anywhere else.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

hyarion, 2025-02-17:
Do I understand you correctly that you are missing time and angle units.

So the following are correct:
* Standard (mm, kg, s, °)
* MKS (m, kg, s, °)

But these:
* US Customary (in, lb)
* FEM (mm, N, s)
* ...

^-- Those are missing some units and should look like this instead:
* US Customary (in, lb, s, °)
* FEM (mm, N, s, °)
* ...

Is that correct?

brendanm250, 2025-02-18:
Yeah that's what I meant, thank you

tobiasfalk, 2025-02-23:
I think it would also be nice if one could switch between deg and rad.

Additionally, if the rad can mode displays it with 1pi, 2pi or 1/2 pi and so.