FreeCAD/FreeCAD #19664: [Core] Actions and commands do not take into account which document is currently active.  (Document isolation)

kaiwas, 2025-02-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The problem is especially relevant when working with two or more documents at the same time.
We can run a command in document 1, then for some reason switch to another document 2 (for example, to look at something in it), while the command will still be active. As if it was launched in document 2, although in fact in document 1.

In addition, if we close a command that was activated "in the neighboring" document - we can get an incomprehensible result. For example, in the video you can see that a parasitic object rendered by a boundibox appears.

A similar problem occurs when switching workbenches, but this is sometimes even useful.

All this is confusing and the user can easily make a mistake if he works with a large amount of information. Probably, a way to "isolate" the document is required, to make it "autonomous". All commands and actions should be active only within the active document. They should NOT appear in running mode when switching documents.

Video demonstration.

https://github.com/user-attachments/assets/74d34743-471c-45a7-8b26-55531d025b4c

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40253 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: e16c462916a050909ad107e4f1ac2e0e5990389d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Logical/physical DPI: 96/91.7938
Installed mods: 
  * FreeCAD_SketchArch
  * PitchedRoof
  * Road 2025.2.6
  * addFC 1.3.0
  * freecad.gears 1.3.0
  * sheetmetal 0.7.11
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-02-17:
> Taking this one step further: Draft and BIM commands do not take the active view into account. For example, it is not possible to pick the start a Draft Line in one view and pick the end in another.

This looks like either a joke or the work of the "external geometry" algorithm (depending on how correctly the online translator translated it for me) :)

Roy-043, 2025-02-17:
I am not joking 😄 . You can open multiple views for the same document, but Draft/BIM commands only works in a single view.

kaiwas, 2025-02-17:
> I am not joking 😄 . You can open multiple views for the same document, but Draft/BIM commands only work in a single view.

The problem is not only in Draft Bim. It concerns absolutely everything. Sketcher, TechDraw, whatever.
if the command is running (any) it remains active when switching between documents.

https://github.com/user-attachments/assets/78be0dd9-d016-418e-8f9f-4bc9ac3867ee

kaiwas, 2025-02-17:
> You can open multiple views

I didn't immediately understand what you were talking about. I've never used multiviews, precisely because I don't see any advantages over the "main" view (including because the commands are locked inside one of several views).

But now I at least understand what you were talking about.

And if you open all the spreadsheets, several documents, techdraw sheets and click "Windows - tiles" everything becomes really interesting. It's completely unclear which window belongs to which document. Everything is dumped in one heap

https://github.com/user-attachments/assets/76defff0-096f-4be8-9afb-fbd741254dd6