FreeCAD/FreeCAD #19665: Unhanced Base: Exception caught in GUIApplication::notify. Error message = Access Violation

ST74650, 2025-02-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Since we have updated Freecad 1.0.0 we cannot use it and we receive the following error message 👍 Unhanced Base: Exception caught in GUIApplication::notify. Error message = Access Violation 

We are under Windows server 2022.

What should we do ?
Regards,
Pierre

### Full version info

```shell
OS: Windows 10 build 20348
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

xtemp09, 2025-02-17:
1) Reproduce the same problem with a weekly bundle
2) If the problem persists, please, report steps to reproduce it.

ST74650, 2025-02-17:
What do you mean : is there a weekly release that I sould implement or is it self managed ?

xtemp09, 2025-02-18:
A weekly bundle is a development release.

https://github.com/FreeCAD/FreeCAD#installing