FreeCAD/FreeCAD #19667: [Draft/Bim] The panel with active layers displays the active layer even if it is not there (when changing the document or closing it).

kaiwas, 2025-02-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When switching the document or closing it, the last selected one is displayed in the panel of active layers. That is, the panel is not updated.

Demonstration on video.

https://github.com/user-attachments/assets/f3583c29-91e2-43f0-b8c5-c6fb3c021dc9

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40253 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: e16c462916a050909ad107e4f1ac2e0e5990389d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Logical/physical DPI: 96/91.7938
Installed mods: 
  * FreeCAD_SketchArch
  * PitchedRoof
  * Road 2025.2.6
  * addFC 1.3.0
  * freecad.gears 1.3.0
  * sheetmetal 0.7.11
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

yorikvanhavre, 2025-02-18:
I guess we need a doc observer here that rebuilds the list when changing the document. But the more tricky part is that it should then remember the state per document, and reapply it on switch.

kaiwas, 2025-02-18:
> the state per document, and reapply it on switch.

It's probably somehow related
https://github.com/FreeCAD/FreeCAD/issues/19664

FEA-eng, 2025-11-04:
@kaiwas Can you confirm if this issue is still relevant?

ect2000, 2026-10-20:
I opened #33769 to remember AutoGroup selections per document and refresh the toolbar on document switches, creation and closing. It includes before/after screenshots and six passing GUI regressions in the native FreeCAD 1.1.4 runner with the changed main-branch Python source. Validation limits and assistance disclosure are included in the PR.