FreeCAD/FreeCAD #19668: Imperial Units

noahdudly, 2025-02-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I know the imperial system sucks, but it sucks a lot more if you are not using the correct format.
I use several CAD programs in my daily life, I Would love to just use metric all day every day.
However, that is not what the customer wants, that is not how the material is supplied, and at certain scales that is not what my brain sees.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Dark/Fusion
Installed mods: 
  * ArchTextures
  * Assembly4 0.50.2
  * Cfd
  * CfdOF 1.25.1
  * freecad.gears 1.0.0
  * kicadStepUpMod 10.22.1
  * Lithophane
  * pcb 6.2023.0
  * Plot 2022.4.17
  * QuickMeasure 2022.10.28
  * Rocket 3.3.0
  * symbols_library
  * ThreadProfile 1.89.0
```

### Subproject(s) affected?

None

### Anything else?

When in the sketcher, for most imperial projects, just use decimal inches, rarely feet, never yards.
When printing a drawing in most cases either FT'IN"1/16 or decimal inches
I know this seems like a rant, and if it is, consider it as frustration that my country has not gone full metric.
Think of what you are measuring, a house, a lawn, a map, or a part:
House = FEET'INCH”1/16th
A boat, even an aircraft carrier is like a house, = FEET'INCH”1/16th or  FEET'INCH.INCH”
A lawn is either FEET'INCH or YARDS yd INCH”
a map is MILES.MILES or like a lawn if the scale is big enough, but not using both in the same scale map.
PARTS, even if they are bigger than 12” are INCH”1/16th OR INCH.INCH” 
If it is for a carpenter the INCH”1/16th if it is for a machinist then INCH.INCH”
Never mix yards and feet.
If the level of precision is more than 1/16”, maybe 1/32” then either use decimal inches or metric.
Fractional measurements always have a divisor that is a power of 2.
Some old people who are way opposed to metric will use smaller fractions, but if it is smaller than a saw kerf you should be using decimal inches anyway. 
Never use decimal feet, or decimal yards, just use inches.
I know the imperial system sucks, but it sucks a lot more if you are not using the correct format.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

WandererFan, 2025-02-18:
Not meaning to be snarky or anything, but what is the actual issue?   Formatting?  The schema themselves?

My opinions on the available non-metric schema:

US Customary is terrible and should be avoided. It should probably be removed.

Imperial Decimal (inch decimal) is good (if you can live with decimals instead of fractions).
Imperial Civil (decimal feet) is also good.

I don't use US Building (feet + fractional inch) so I can't comment.  

None of the wood products I can buy locally actually measure out to their nominal Imperial size, so I use Metric Small Parts (mm) for woodworking.

maxwxyz, 2025-02-22:
@noahdudly you can reopen the issue when you rephrased it to be an actionable issue. TIA