FreeCAD/FreeCAD #19669: BIM: When active Arch_SectionPlane is deleted, the cut stays active in 3D view

semhustej, 2025-02-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When Arch_SectionPlane is deleted with `CutView` = `True`, the cut stays active in 3D view after the deletion. I tried to recompute the whole document with no positive result. 

Steps to reproduce:

- open sample file 

[test-delete-section-plane.FCStd.zip](https://github.com/user-attachments/files/18827944/test-delete-section-plane.FCStd.zip)

- delete the Section object
- the cut is active in 3D view with no apparent option for user to remove it

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40239 (Git) Conda
Build type: Release
Branch: main
Hash: 3623f8baf50f924fd2a213495c72a5e316cf9341
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073
```

### Subproject(s) affected?

BIM

### Anything else?

Workaround: close and re-open the document.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ErtugrulEge, 2025-02-17:
Please, check your zip file.I couldn't find the FC file  but I created walls and a section plane and I can confirm the issue.

furgo16, 2025-02-17:
The attached file is fine, you only need to rename it to remove the `.zip` suffix. I can confirm as well:

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40267 (Git) Snap 1303
Build type: Release
Branch: main
Hash: 04d2bcbf45ab96f3895ac3d0e8b5ae9becbaf5cd
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 7.1.1, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/159.34
```

yorikvanhavre, 2025-02-18:
Indeed I don't think anything is made for that case... It probably needs a `def unsetupObject(self, obj)` that removes the clip plane if it's present

semhustej, 2025-05-21:
Another report facing this issue: https://forum.freecad.org/viewtopic.php?p=828002

ErtugrulEge, 2025-11-28:
Still relevant
```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20251126 (Git shallow)
Build date: 2025/11/26 00:30:17
Build type: Release
Branch: (HEAD detached at de230cf)
Hash: de230cfaa75433441366ad09a4d5fda828e7bfdb
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * CurvedShapes 1.0.13
  * Curves 0.6.61
```