FreeCAD/FreeCAD #19677: The root directory of the user Mod is not in sys.path

uyoufu, 2025-02-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Because The root directory of the user Mod (C:\\Users\\galens\\AppData\\Roaming\\FreeCAD\\Mod) is not in `sys.path`, I can not import custom Mod as python package, ex framework Mod

`sys.path` result:
```
[
  "C:\\Users\\uyoufu\\AppData\\Roaming\\FreeCAD\\Mod\\Road\\.\\",
  "C:\\Users\\uyoufu\\AppData\\Roaming\\FreeCAD\\Mod\\Road",
  "C:\\Users\\uyoufu\\AppData\\Roaming\\FreeCAD\\Mod\\Geomatics",
  "C:\\Users\\uyoufu\\AppData\\Roaming\\FreeCAD\\Mod\\geodata",
  "C:\\Users\\uyoufu\\AppData\\Roaming\\FreeCAD\\Mod\\framework",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\Web",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\Tux",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\Test",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\TechDraw",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\Surface",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\Start",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\Spreadsheet",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\Sketcher",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\Show",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\Robot",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\ReverseEngineering",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\Points",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\Plot",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\PartDesign",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\Part",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\OpenSCAD",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\MeshPart",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\Mesh",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\Measure",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\Material",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\Inspection",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\Import",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\Idf",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\Help",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\Fem",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\Draft",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\CAM",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\BIM",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\Assembly",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod\\AddonManager",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Mod",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\lib",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Ext",
  "D:\\scoop\\apps\\freecad-weekly\\current\\bin\\python311.zip",
  "D:\\scoop\\apps\\freecad-weekly\\current\\bin\\DLLs",
  "D:\\scoop\\apps\\freecad-weekly\\current\\bin\\Lib",
  "D:\\scoop\\apps\\freecad-weekly\\current\\bin",
  "D:\\scoop\\apps\\freecad-weekly\\current\\bin\\Lib\\site-packages",
  "C:\\Users\\uyoufu\\AppData\\Roaming\\FreeCAD\\Macro\\",
  "C:\\Users\\uyoufu\\AppData\\Roaming\\FreeCAD\\Macro",
  "D:\\scoop\\apps\\freecad-weekly\\40176\\Macro",
];

```

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40176 (Git) Conda
Build type: Release
Branch: main
Hash: 2745f436026b6f1fb84dbaaf89ea22b5fb4c2295
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: Chinese/China (zh_CN)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/93.0817
Installed mods: 
  * framework
  * geodata
  * Geomatics
  * Road 2025.2.6
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-02-18:
@uyoufu If you haven't got the build from the previous week to prove that the regression is very recent, could you please test with the 40041 build from [here](https://github.com/kisolre/FreeCAD/releases/tag/weekly_Win7)

uyoufu, 2025-02-19:
I test the version 40303 build from [here](https://github.com/FreeCAD/FreeCAD-Bundle/releases/) now, bug still exists

``` python
import sys
sys.path
```

Syres916, 2025-02-19:
I specifically asked you to test the older build to help reduce the time required to trace the commit that caused the issue, I know it's not been fixed. Anyway I'll just do it all myself.

Syres916, 2025-02-19:
This isn't a regression, it's been like this on Windows for years.

If `framework` is a proper Python package like `ezdxf` then it should go into:
C:\\Users\\uyoufu\\AppData\\Roaming\\FreeCAD\\AdditionalPythonPackages\\py311 (folders which may need creating).

If `framework` is a workbench that you are creating yourself then it must not have all the correct files in it's root directory, use `Road` as a good template for what needs to be in `package.xml` and `init_gui.py`

closed 2025-02-19 by maxwxyz