FreeCAD/FreeCAD #19681: PPA: daily build failure 2025-02-12

furgo16, 2025-02-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The daily builds have been failing for about a week, since 2025-02-12: [2025-02-12](https://launchpad.net/~freecad-maintainers/+archive/ubuntu/freecad-daily/+packages?field.name_filter=&field.status_filter=&field.series_filter=noble)

There's a forum thread that points to this PR as the cause, but I've not confirmed it myself.

Forum: https://forum.freecad.org/viewtopic.php?style=2&t=94820

### Full version info

```shell
N/A
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-02-18:
@hyarion @adrianinsaval any ideas about the cause of the failure? Thanks.

hyarion, 2025-02-19:
Thanks for the report.

To my understanding the compiler should be able to see that:
1. this `static_assert` shouldn't be evaluated in this case
2. the `static_assert` is fulfilled in these cases

That said, I do recognize that we need to find a workaround this issue.
I see two ways forward:
* we can ifdef the compiler type/version as a workaround
* try to see if `consteval` in c++20 solves this issue.

@furgo16 do you know if there is a way to reproduce the error in a container build? (it's difficult to find a solution without being able to test it locally)

hyarion, 2025-02-19:
@maxwxyz feel free to assign me to the issue

furgo16, 2025-02-19:
Thanks! I could only think of doing a container build with the same `CMAKE_CXX_FLAGS` as the .deb build, as that seems to be the main difference.

furgo16, 2025-02-19:
This is a summary of the repo locations and how the PPA build works, at least to my understanding: https://github.com/FreeCAD/FreeCAD-Bundle/issues/354#issuecomment-2537111118

hyarion, 2025-02-23:
I haven't forgot about his, I'm just waiting for c++20 to hit the main branch.

furgo16, 2025-02-24:
No worries, I'm watching https://github.com/FreeCAD/FreeCAD/pull/19725 as well.

furgo16, 2025-02-27:
@hyarion with https://github.com/FreeCAD/FreeCAD/pull/19725 having been merged, would you have some time to look at this issue again? Thanks!

mfraser, 2025-03-12:
Looks like this may be fixed again as I just had an update from the PPA.
```
OS: Ubuntu 24.04.2 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40675 (Git)
Build date: 2025/03/11 12:58:03
Build type: Release
Branch: main
Hash: 61eedbddc4dfc9416995262e4ad182078bfbab02
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 100.911/102.706/1
```

hyarion, 2025-03-12:
The discussion moved to the PR linked above.
But thanks for confirming that it got built!

furgo16, 2025-03-12:
Indeed, https://github.com/FreeCAD/FreeCAD/pull/20049 fixes the issue. Thanks everyone!

closed 2025-03-12 by furgo16