FreeCAD/FreeCAD #19686: Empty Assembly file after crash (despite saving).

Stutchbury, 2025-02-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Assembly v1.0 crashes frequently. Very frequently. I'm attempting to model just 6 Revolute joint and one Cylindrical joint. 

Unfortunately on about 1 in 10 crashes, when re-opening a previously saved assembly FCstd file, it is empty. Fortunately my model if fairly easily recreated but I would imagine this would be most unfortunate on a large model.

This is the message on re-opening:

`
16:37:12  Fatal Error at file /path/ottoman-mechanism-assembly.FCStd, line 151, char 27
16:37:12  Invalid Document.xml: unexpected end of input
16:37:12  Fatal Error at file GuiDocument.xml, line 1, char 1
16:37:12  Exception message is: 
16:37:12  invalid document structure
16:37:12  Reading failed from embedded file: GuiDocument.xml (0 bytes, 2 bytes compressed)

`

This has occurred in both v1.0 and v1.1dev (I am using separate files for each assembly with the same v1.0 base files but am not saving the external files).

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.40
  * kicadStepUpMod 11.2.4

--

OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.1.0dev.40253 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: e16c462916a050909ad107e4f1ac2e0e5990389d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/157.74
Installed mods: 
  * Curves 0.6.40
  * kicadStepUpMod 11.2.4
```

### Subproject(s) affected?

Assembly

### Anything else?

Attached empty file:

[ottoman-mechanism-assembly.zip](https://github.com/user-attachments/files/18849131/ottoman-mechanism-assembly.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-22:
Please add steps how to reproduce a crash

drwho495, 2025-02-23:
Is 1.1dev crashing on files created in 1.1 or 1.0

Stutchbury, 2025-02-23:
> Is 1.1dev crashing on files created in 1.1 or 1.0

The Parts were created in 1.0 and the crashes occurred in 1.0. I also tested with 1.1dev but have been very careful not to 'Save External Files'.

drwho495, 2025-02-23:
> > Is 1.1dev crashing on files created in 1.1 or 1.0
> 
> The Parts were created in 1.0 and the crashes occurred in 1.0. I also tested with 1.1dev but have been very careful not to 'Save External Files'.

Try it on files created in 1.1dev

Stutchbury, 2025-02-23:
> Please add steps how to reproduce a crash

1. Open a new file and create an assembly
2. Import 4 parts and connect ends with Revolut joint (kinda like a rectangle)
3. Import a tube and a rod (to become a gas strut) and connect with Revolut and Cylindrical joints.
4. Move things around and (I think) lose focus on the Assembly, double click on Assembly to reactivate and crash.

Step 4 is what just happened, but couldn't say that was always the trigger. This time did not lose the contents of the file.

Stutchbury, 2025-02-23:
> > > Is 1.1dev crashing on files created in 1.1 or 1.0
> > 
> > 
> > The Parts were created in 1.0 and the crashes occurred in 1.0. I also tested with 1.1dev but have been very careful not to 'Save External Files'.
> 
> Try it on files created in 1.1dev

OK, will do.

luzpaz, 2025-09-01:
@Stutchbury ?

Stutchbury, 2025-09-01:
My apologies for the tardy response - I think I had likely been saving files across versions and ended up with a large number of differently named versions. Much confusion followed so I deleted them all and started again - very careful to not save across versions (the original project was a pre-v1.0 one).
Please close this as things have likely changed somewhat.

closed 2025-09-01 by Stutchbury