FreeCAD/FreeCAD #19689: Issue with FreeCAD Dark Theme and Part Design Modeling Functions

AndroideData, 2025-02-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I've encountered an issue with the FreeCAD Dark theme. When using this theme, the "Part Design Modeling" functions do not work correctly. I receive the following error message: "Cannot use selected object. Selected object must belong to the active body. Consider using a ShapeBinder or a BaseFeature to reference external geometry in a body."

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Spanish/Spain (es_ES)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-02-19:
I've looked the error message up in the code and it's generated when the user is trying to use a sketch/profile that is not in the active Body and this has absolutely no effect of the FreeCAD Dark Theme, this should be removed from the title of this issue as it's completely irrelevant. If you follow the [Wiki](https://wiki.freecad.org/PartDesign_Workbench) regarding Part Design Wb and if you believe there's not sufficient documentation regarding this please provide a suggested text to replace.

AndroideData, 2025-02-19:
Thank you for your response. I understand that the error is unrelated to the Dark Theme. To clarify my issue further, I've created a video demonstrating the problem I'm encountering. Please take a look:

https://youtu.be/qE0SZuuDunA

If possible, could you guide me on what might be causing this issue and how I can resolve it? Any additional suggestions would be greatly appreciated.

Thank you for your assistance.

maxwxyz, 2025-02-19:
Please use the forum for help requests.

closed 2025-02-19 by maxwxyz