FreeCAD/FreeCAD #19692: Font size for dimensions won't change

jtg909, 2025-02-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I can't get the font size for dimensions to change - the dimensions labels are tiny for any font size up to 11 and don't change, anything over 11cm the label disappears entirely. See attached image. I was following the tutorial here: https://www.youtube.com/watch?v=IV-nv9ygZPM&t=1666s at minute 48:54. I didn't have the large font that the tutorial describes as "the elephant in the room" just a tiny label when I reduced the size to 0.5cm

[image]

I am using the TechDraw workbench

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

benj5378, 2025-02-22:
I cannot reproduce either.

@jtg909 what am I supposed to see on your screenshot? I see a fontsize set to 0,5 cm, and the visual font size looking like 0,5 cm, as expected?

Syres916, 2025-02-25:
Is there a reason why the font being used isn't the supplied `osifont` or at the very least a proper TrueType font?

benj5378, 2025-03-29:
If there are no new updates on this, let's close

closed 2025-03-29 by WandererFan

Ernurcum, 2025-07-09:
import FreeCADGui as Gui

# Separate stylesheet strings
qss_all = """
* {
    font-size: 15pt;
}
"""

qss_qdialog = """
QDialog {
    font-size: 10pt;
}
"""

qss_qlabel = """
QLabel {
    font-size: 10pt;
}
"""

qss_qpushbutton = """
QPushButton {
    font-size: 10pt;
}
"""

qss_qlineedit = """
QLineEdit {
    font-size: 15pt;
}
"""

# Combine all styles
qss_combined = qss_all + qss_qdialog + qss_qlabel + qss_qpushbutton + qss_qlineedit

# Apply to GUI
Gui.getMainWindow().setStyleSheet(qss_combined)
print("Custom font stylesheet applied to GUI")