FreeCAD/FreeCAD #19694: Dropdown for Attachment mode in Datum Plane doesn't go away

Sjorsa, 2025-02-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When clicking the Datum Plane  button, the properties ask for an Attachment mode. Doesn't matter what mode I select, the dropdown just stays up. I don't know if there's anything behind it. Doesn't feel very intuitive.

### Full version info

```shell
OS: KDE Flatpak runtime (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-22:
Which dropdown? Could you add screenshots and steps how to reproduce?

Sjorsa, 2025-03-04:
Excuse me for the late response. After looking back I think I just viewed it wrong. It looked like the Attachment Mode was a dropdown that was hiding stuff behind it, but the dropdown just isn't supposed to collapse. It is probably my theme that made it look confusing to me, since the screenshot in the [documentation](https://wiki.freecad.org/Part_EditAttachment) makes it look more clear.

I'll close this issue now

[image]

closed 2025-03-04 by Sjorsa