FreeCAD/FreeCAD #19698: VarSet equation is not returning the right value.

leoheck, 2025-02-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

See the `Tube > Ext_Diameter`. There is an equation with the resulting value `0.0` which is wrong.
[image]

Opening the equation I can see the right value tho, which is `7.86`

[image]

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40303 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 145af5cddc9774f00fd4dfb3bd35e94efc5c70ca
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/145.143
Installed mods: 
  * Assembly4 0.50.18-dev
  * sheetmetal 0.7.5
  * Curves 0.6.54
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * GDT
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.26
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leoheck, 2025-02-19:
If I add 1, it returns the right value.

[image]

leoheck, 2025-02-19:
Changing the order or the tan makes it work.
[image]

kaiwas, 2025-02-19:
The problem only occurs when creating the first property.
Until the moment of recalculation. If you press F5, the data is updated.

https://github.com/user-attachments/assets/8868f06a-eff3-418b-a38f-5436539348ae

All subsequent properties that we add are recalculated automatically without pressing F5

maxwxyz, 2025-02-19:
@pieterhijma could you please take a look at this?

leoheck, 2025-02-19:
I am thinking this happening since I am using another variable set in the VarSet (`height`).

pieterhijma, 2025-03-17:
> [@pieterhijma](https://github.com/pieterhijma) could you please take a look at this?

Yes, I will fix this. I think it is a small problem with the dialog: If you enter an expression, could be as simple as "2", then the dialog shows the correct value, but the value is not transferred yet to the property as well, so you see something like "0.00 mm ( 2 )" in the property view. It would be better if this value is actually set. It's a small problem because luckily the VarSet shows that it should refresh itself and if that happens, the value is correctly updated.

closed 2025-07-25 by kadet1090