FreeCAD/FreeCAD #19702: Fillet on curve surface or edge will crash

Wilson1972-0815, 2025-02-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I select a edge(or face) and try to add Fillet. When 'Fillet' button hit, the application crash.

This my .FCStd
[ArcheryArmGuard.FCStd.zip](https://github.com/user-attachments/files/18863634/ArcheryArmGuard.FCStd.zip)

### Full version info

```shell
FreeCAD 1.0.0. I don't where to copy information from "About FreeCAD dialog". Run on Mac.
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tarman3, 2025-02-19:
confirm

<details>
  <summary>FreeCAD 1.1.0dev.40303 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40303 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 145af5cddc9774f00fd4dfb3bd35e94efc5c70ca
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/118.293
Installed mods: CurvedShapes 1.0.13; lattice2 1.0.0; Silk 0.1.5; freecad.gears 1.3.0; btl 0.9.9 (Disabled); fasteners 0.5.37; MeshRemodel 1.10.36; Curves 0.6.58; sheetmetal 0.7.11; Defeaturing 1.2.3

```
</details>

pieterhijma, 2025-03-13:
Can reproduce with 8.0.0dev as well:

```
OS: Arch Linux (xcb)
Architecture: x86_64
Version: 1.1.0dev.40717 (Git)
Build date: 2025/03/13 13:32:18
Build type: Release
Branch: main
Hash: a375752aa4ea4d416f757299c716820546fd41cc
Python 3.13.2, Qt 6.8.2, Coin 4.0.3, Vtk , OCC 8.0.0.dev-3840d19d7b
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1.33333
Installed mods: 
  * OpenTheme 2024.5.16
  * Ondsel-Lens 2024.10.21.01
  * fasteners 0.5.34
  * WebTools 1.0.0
  * osh-autodoc-workbench 0.2.3
  * DynamicData 2.70.0
```

<details>
<summary>Stacktrace:</summary>
<pre>
#0  0x00007fff7727be8c in opencascade::handle<TopLoc_SListNodeOfItemLocation>::handle (this=0x7fffffff6e58, theHandle=...) at /home/pieter/current/occt-liaison/occt-main/src/Standard/Standard_Handle.hxx:73
#1  0x00007fff7727bb91 in TopLoc_SListOfItemLocation::TopLoc_SListOfItemLocation (this=0x7fffffff6e58, Other=...) at /home/pieter/current/occt-liaison/occt-main/src/TopLoc/TopLoc_SListOfItemLocation.hxx:57
#2  0x00007fff7727be43 in TopLoc_Location::TopLoc_Location (this=0x7fffffff6e58) at /home/pieter/current/occt-liaison/occt-main/src/TopLoc/TopLoc_Location.hxx:35
#3  0x00007fff7727ae34 in TopLoc_Location::Multiplied (this=0x7fffffff70a8, Other=...) at /home/pieter/current/occt-liaison/occt-main/src/TopLoc/TopLoc_Location.cxx:98
#4  0x00007fff7a515dec in TopLoc_Location::operator* (this=0x7fffffff70a8, Other=...) at /home/pieter/current/occt-liaison/occt-main/src/TopLoc/TopLoc_Location.hxx:97
#5  0x00007fff7a53b262 in BRep_Tool::Surface (F=..., L=...) at /home/pieter/current/occt-liaison/occt-main/src/BRep/BRep_Tool.cxx:79
#6  0x00007fff7a554a62 in BRepAdaptor_Surface::Initialize (this=0x55555f4bc3b0, F=..., Restriction=true) at /home/pieter/current/occt-liaison/occt-main/src/BRepAdaptor/BRepAdaptor_Surface.cxx:75
#7  0x00007fff80525f23 in ChFi3d_Builder::StartSol (this=0x7fffffff8ce0, Spine=..., HS=..., pons=..., HC=..., W=@0x7fffffff7410: 18.6185155244901, SD=..., isfirst=false, ons=2, HSref=..., HCref=..., RecP=@0x7fffffff7362: false, RecS=@0x7fffffff7363: false, RecRst=@0x7fffffff7364: false, c1obstacle=@0x7fffffff7365: false, HSBis=..., PBis=..., decroch=false, Vref=...) at /home/pieter/current/occt-liaison/occt-main/src/ChFi3d/ChFi3d_Builder_2.cxx:1199
#8  0x00007fff8052c435 in ChFi3d_Builder::PerformSetOfSurfOnElSpine (this=0x7fffffff8ce0, HGuide=..., Stripe=..., It1=..., It2=..., Simul=false) at /home/pieter/current/occt-liaison/occt-main/src/ChFi3d/ChFi3d_Builder_2.cxx:2420
#9  0x00007fff80530b01 in ChFi3d_Builder::PerformSetOfKGen (this=0x7fffffff8ce0, Stripe=..., Simul=false) at /home/pieter/current/occt-liaison/occt-main/src/ChFi3d/ChFi3d_Builder_2.cxx:3215
#10 0x00007fff80533f6c in ChFi3d_Builder::PerformSetOfSurf (this=0x7fffffff8ce0, Stripe=..., Simul=false) at /home/pieter/current/occt-liaison/occt-main/src/ChFi3d/ChFi3d_Builder_2.cxx:3748
#11 0x00007fff804eacc4 in ChFi3d_Builder::Compute (this=0x7fffffff8ce0) at /home/pieter/current/occt-liaison/occt-main/src/ChFi3d/ChFi3d_Builder.cxx:258
#12 0x00007fff8068d580 in BRepFilletAPI_MakeFillet::Build (this=0x7fffffff8c90) at /home/pieter/current/occt-liaison/occt-main/src/BRepFilletAPI/BRepFilletAPI_MakeFillet.cxx:375
#13 0x00007fff78482a80 in BRepBuilderAPI_MakeShape::Shape (this=0x7fffffff8c90) at /home/pieter/current/occt-liaison/occt-main/src/BRepBuilderAPI/BRepBuilderAPI_MakeShape.cxx:35
#14 0x00007fff7bc6c111 in Part::TopoShape::makeElementShape (this=0x7fffffff93c0, mkShape=..., shapes=std::vector of length 1, capacity 1 = {...}, op=0x7fff7bd41d0e "FLT") at /home/pieter/current/occt-liaison/freecad-occt-main/src/Mod/Part/App/TopoShapeExpansion.cpp:4145
#15 0x00007fff7bc6c024 in Part::TopoShape::makeElementShape (this=0x7fffffff93c0, mkShape=..., source=..., op=0x7fff7bd41d0e "FLT") at /home/pieter/current/occt-liaison/freecad-occt-main/src/Mod/Part/App/TopoShapeExpansion.cpp:4132
#16 0x00007fff7bc6a501 in Part::TopoShape::makeElementFillet (this=0x7fffffff93c0, shape=..., edges=std::vector of length 1, capacity 1 = {...}, radius1=1, radius2=1, op=0x7fff7bd41d0e "FLT") at /home/pieter/current/occt-liaison/freecad-occt-main/src/Mod/Part/App/TopoShapeExpansion.cpp:4008
#17 0x00007fff6ab4714d in PartDesign::Fillet::execute (this=0x55555d86f630) at /home/pieter/current/occt-liaison/freecad-occt-main/src/Mod/PartDesign/App/FeatureFillet.cpp:98
#18 0x00007ffff3cbc58e in App::DocumentObject::recompute (this=0x55555d86f630) at /home/pieter/current/occt-liaison/freecad-occt-main/src/App/DocumentObject.cpp:172
#19 0x00007fff7b8ce9e7 in Part::Feature::recompute (this=0x55555d86f630) at /home/pieter/current/occt-liaison/freecad-occt-main/src/Mod/Part/App/PartFeature.cpp:107
#20 0x00007fff6aa3b819 in PartDesign::Feature::recompute (this=0x55555d86f630) at /home/pieter/current/occt-liaison/freecad-occt-main/src/Mod/PartDesign/App/Feature.cpp:82
#21 0x00007ffff3b82599 in App::Document::_recomputeFeature (this=0x55555b9456e0, Feat=0x55555d86f630) at /home/pieter/current/occt-liaison/freecad-occt-main/src/App/Document.cpp:3345
#22 0x00007ffff3b7fd5c in App::Document::recompute (this=0x55555b9456e0, objs=std::vector of length 0, capacity 0, force=false, hasError=0x0, options=0) at /home/pieter/current/occt-liaison/freecad-occt-main/src/App/Document.cpp:3055
[... cut ...]
#78 0x00007ffff58530c9 in (anonymous namespace)::tryRunEventLoop (mainApp=...) at /home/pieter/current/occt-liaison/freecad-occt-main/src/Gui/Application.cpp:2230
#79 0x00007ffff5853423 in (anonymous namespace)::runEventLoop (mainApp=...) at /home/pieter/current/occt-liaison/freecad-occt-main/src/Gui/Application.cpp:2259
#80 0x00007ffff58537dc in Gui::Application::runApplication () at /home/pieter/current/occt-liaison/freecad-occt-main/src/Gui/Application.cpp:2339
#81 0x000055555557cc36 in main (argc=1, argv=0x7fffffffd468) at /home/pieter/current/occt-liaison/freecad-occt-main/src/Main/MainGui.cpp:313
</pre>
</details>

pieterhijma, 2025-04-10:
I'm currently looking into this. This may not be an OCC issue, but could also be a FreeCAD issue. In any case, there is something strange in the model that I don't understand: There is a Cut that contains a Common. This Common is created from a Pad and from an Extrude. The Pad is no longer in the Body but it seems to be "eaten up" by Common. I'm going to try to reproduce this and create a separate issue for it.

Perhaps @kadet1090, since you have been looking into uniting Part and Part Design, is this something you know anything about?

pieterhijma, 2025-04-14:
To pinpoint the problem, I need to make a minimal model with which it crashes. With this minimal model I can create an issue that I can send to OpenCascade. This comment describes this process:

- When you remove Fillet001, the issue still persists. 
- When you remove Fillet, the issue still persists.

To reproduce the issue, I select this edge:

[image]

This edge is part of PadSpine, but if I remove PadSpine, the Body is more or less empty:

[image]

SketchSpine is based on Cut, so if I show Cut and select the edge from that and try `PartDesign_Fillet`, I get:

[image]

I believe this is the core problem of this issue. PartDesign bodies are based on Part features and vice versa. 

If I do `Part_Fillet` on Cut, it works fine:

[image]

If I do a `Part_Fillet` on the original model (so Fillet001 in Body), FreeCAD crashes as well.

For now, my conclusion is that in any case this is a FreeCAD problem related to the interaction between Part and PartDesign features. It may still be an OCC problem, but from the perspective of the OCC Liaison position, it does not make much sense to investigate this further and report it to OCCT before the interaction between Part and PartDesign has been solved.

I'll refer to this comment from https://github.com/FreeCAD/FreeCAD/issues/20735 and will mark this as Done in the OCCT Liaison project.