FreeCAD/FreeCAD #19703: Unexpected result from fillet

Sherman-c, 2025-02-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Tried to create fillet with 1mm between 3mm cylinder and 3mm surface, forgive my English, it is a bit hard for me to explain, but please see attached files, everything was ok until I added the edge on the other side, it looks pretty odd.

[image]

[image]

[image]

 Regards
Sherman

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * A2plus 0.4.68
  * Assembly4 0.50.15
  * Curves 0.6.53
  * fcgear 1.0.0
  * lattice2 1.0.0
  * Lithophane
  * MeshRemodel 1.10.35
  * OpticsWorkbench 1.0.26
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Sherman-c, 2025-02-20:
Hi tarman3
It appear I'm not allow to upload/attach and freecad file, tried FCStd/3MF and STL, all those file types are blocked.  I rename the file from Arrow.FCStd to arrow.txt and upload it.  you may need to rename it back before open it in freecad

The file contains 2 bodies, the one I have problem with is the body named head

[arrow.txt](https://github.com/user-attachments/files/18896369/arrow.txt)

Thanks
Sherman

Kreyren, 2025-02-22:
From https://github.com/FreeCAD/FreeCAD/issues/19675 for context

[image]

luzpaz, 2026-04-25:
@Sherman-c is it fixed in 1.1.0 ? Or did you have to do a workaround ?  
Please also provide full [About](https://wiki.freecad.org/About) info. Thanks in advance

Edit: ping @Kreyren (please test as well, thank you.)