FreeCAD/FreeCAD #19704: CAM: Cannot assign tools and stock material to g-code loaded from file

Harvie, 2025-02-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hello,
when i open pre-existing gcode file in CAM workbench, it is correctly loaded and displayed in FreeCAD. But there is no way to assign tool table and stock material to it, which i beleive is the only reason why g-code imported from external file cannot be simulated. (as opposed to paths generated directly in FreeCAD project, which obviously do have tools and stock material assigned).

### Full version info

```shell
OS: Arch Linux (XFCE/Xfce Session/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.13.1, Qt 6.8.2, Coin 4.0.3, Vtk 9.4.1, OCC 7.8.1
Locale: Czech/Czechia (cs_CZ)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tarman3, 2025-02-19:
Please attach FCStd file

tarman3, 2025-02-19:
You should create model, stock and **Job** manually
Create **Custom** operation and inset gcode without start 3 lines
After that you can run simulation

[bush.zip](https://github.com/user-attachments/files/18873063/bush.zip)

[image]

maxwxyz, 2025-02-19:
If there is still a bug. please reopen. You can use the forum for help requests.

closed 2025-02-19 by maxwxyz

Harvie, 2025-02-19:
My bad, i need to figure out what i was doing wrong. Thank you.

Harvie, 2025-02-28:
It seems copypasting into custom operation rather than opening the g-code file directly did the trick. but the toolchanges seem to be ignored as the custom operation has hard-assigned tool. (also M6 T2 toolchange command breaks the new g-code simulator, so i had to remove it as you suggested).

Also is there a way to create stock without having model first? It would be nice to have some kind of automaticaly generated stock based on g-code, just for simulation purposes, when there is no model.

Harvie, 2025-02-28:
I've made tutorial video about simulating external g-code in freecad:
https://www.youtube.com/watch?v=6qRtgYtHp94