FreeCAD/FreeCAD #19707: PartDesign Helix - Turns should display more digits.

leoheck, 2025-02-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

`Turns` should show more digits.

Turns are like a percentage. I have a big thread diameter. I want to have just a bit of a turn.

[image]

I am trying to model this thread stop at the end of the thread.

### Full version info

```shell
1.1-dev
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-19:
@FlachyJoe is this possible or related to the decimal setting in the preferences?

leoheck, 2025-02-19:
For instance, in this example, I am using 0.001