FreeCAD/FreeCAD #19709: Crash on when clic on measure label only on DXF import

glt248, 2025-02-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

1. Import a DXF
2. Use Measure tool to get circle radius
3. Click on the measure label

Here is the log
```bash
FreeCAD_1.0.0-conda-Linux-x86_64-py311.AppImage 
FreeCAD 1.0.0, Libs: 1.0.0R39109 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

/home/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/linux/AppDir/usr/share/libdrm/amdgpu.ids: Aucun fichier ou dossier de ce nom
No INSUNITS or MEASUREMENT; setting scaling to 1 DXF unit = 1mm based on DXF scaling option
First reference to missing Layer '0'First reference to missing Layer '29'First reference to missing Layer '30'Unsupported DXF features:
Entity type 'SOLID': 3 time(s) first at line 138
No measure handler available for geometry of module: Measure
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3fda0) [0x7f9fae449da0]
#1  0x7f9fb1885d43 in Gui::SoFCUnifiedSelection::setSelection(std::vector<Gui::SoFCUnifiedSelection::PickedInfo, std::allocator<Gui::SoFCUnifiedSelection::PickedInfo> > const&, bool) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/libFreeCADGui.so+0x613
#2  0x7f9fb18873db in Gui::SoFCUnifiedSelection::handleEvent(SoHandleEventAction*) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/libFreeCADGui.so+0x32b
#3  0x7f9fadcd54e8 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/./libCoin.so.80+0x68
#4  0x7f9fade8cb9a in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/./libCoin.so.80+0xea
#5  0x7f9fadedc348 in SoGroup::doAction(SoAction*) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/./libCoin.so.80+0x58
#6  0x7f9fadef713a in SoSeparator::doAction(SoAction*) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/./libCoin.so.80+0x2a
#7  0x7f9fadcd54e8 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/./libCoin.so.80+0x68
#8  0x7f9fadcdf26c in SoHandleEventAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/./libCoin.so.80+0xac
#9  0x7f9fadcd61ba in SoAction::apply(SoNode*) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/./libCoin.so.80+0x2ea
#10  0x7f9fadea8ad2 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/./libCoin.so.80+0x72
#11  0x7f9fb193714e in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/libFreeCADGui.so+0x1e
#12  0x7f9fb1942618 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/libFreeCADGui.so+0x58
#13  0x7f9fb194c360 in Gui::NavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/libFreeCADGui.so+0x40
#14  /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/libFreeCADGui.so(+0x95d219) [0x7f9fb195d219]
#15  /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/libFreeCADGui.so(+0x95d5f9) [0x7f9fb195d5f9]
#16  0x7f9fb195b9c2 in Gui::GestureNavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/libFreeCADGui.so+0x2a2
#17  0x7f9fb194e265 in Gui::NavigationStyle::processEvent(SoEvent const*) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/libFreeCADGui.so+0x65
#18  0x7f9fb1977c3b in Gui::View3DInventorViewer::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/libFreeCADGui.so+0x6b
#19  0x7f9fb192d9e5 in SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/libFreeCADGui.so+0x1a5
#20  0x7f9fae8b68bb in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/libQt5Core.so.5+0x7b
#21  0x7f9faf77039e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/libQt5Widgets.so.5+0x6e
#22  0x7f9faf7778f9 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/libQt5Widgets.so.5+0xa39
#23  0x7f9fb1569559 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/libFreeCADGui.so+0xc9
#24  0x7f9fae8b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/libQt5Core.so.5+0x108
#25  0x7f9faf7763eb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#26  /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/libQt5Widgets.so.5(+0x1c6e6f) [0x7f9faf7c6e6f]
#27  /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca0c9) [0x7f9faf7ca0c9]
#28  0x7f9faf7703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#29  0x7f9fb1569559 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/libFreeCADGui.so+0xc9
#30  0x7f9fae8b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/libQt5Core.so.5+0x108
#31  0x7f9faef353ec in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/libQt5Gui.so.5+0x84c
#32  0x7f9faef0b0bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/libQt5Gui.so.5+0xac
#33  /tmp/.mount_FreeCAbCfhPL/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x7f9fa9394bea]
#34  /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x7f9facd10ad7]
#35  /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x7f9facd13ac7]
#36  /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7f9facd140ce]
#37  0x7f9fae90aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/libQt5Core.so.5+0x66
#38  0x7f9fae8b570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/libQt5Core.so.5+0x11b
#39  0x7f9fae8bd212 in QCoreApplication::exec() from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/libQt5Core.so.5+0x82
#40  0x7f9fb14c8744 in Gui::Application::runApplication() from /tmp/.mount_FreeCAbCfhPL/usr/bin/../lib/libFreeCADGui.so+0x8a4
#41  /tmp/.mount_FreeCAbCfhPL/usr/bin/freecad(+0x78b7) [0x5559a50368b7]
#42  /lib/x86_64-linux-gnu/libc.so.6(+0x29d68) [0x7f9fae433d68]
#43  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x85) [0x7f9fae433e25]
#44  /tmp/.mount_FreeCAbCfhPL/usr/bin/freecad(+0x7c2a) [0x5559a5036c2a]
```

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (XFCE/lightdm-xsession/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * kicadStepUpMod 11.5.2
```

### Subproject(s) affected?

None

### Anything else?

Than you for nice V1 !

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

CaptainCoward, 2025-02-24:
I can confirm this crash. I just experimented a bit with freecad so my example file is pretty much chaos - since i don't have much clue what I'm actually doing. Though you can reproduce the crash pretty frequently there. Just click measurement, measure a few spots  and keep clicking into the measurement label like on my screenshot. 

[image]

[base.zip](https://github.com/user-attachments/files/18949677/base.zip)

ErtugrulEge, 2025-04-06:
If you click the label,still gives some error without a dxf import

[image]

ErtugrulEge, 2025-08-19:
I am getting `Selection.cpp(1635): Object not found ` error when click label on dev version.

https://github.com/user-attachments/assets/00a329f9-bd77-48ef-8662-4806b13ff514

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/12 22:08:35
Build type: Release
Branch: main
Hash: 4195f7fee28e65b289b9390ff7a0deab49e81040
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * CurvedShapes 1.0.13
  * Curves 0.6.61
```

FEA-eng, 2025-09-29:
I don't have a crash on Windows but indeed there's an error and the measurement disappears when clicking on any label (doesn't matter if it's DXF or not)

`10:50:10  0 Selection.cpp(1634): Object not found`

chennes, 2025-10-02:
What's happening with the "Object not found" message is that you are in measure mode, and you click on a label that is *not* on top of something that can be measured. Since it makes no sense to measure a label, the code gives you a message that whatever you thought you were clicking on was not a measurable object (which is true).

So there are a couple questions we have to answer before moving forward with this bug report:
1. Is there still a crash in here someplace? As far as I can tell, nothing is crashing in 1.1dev. If you have an example DXF and some instructions for how to make FreeCAD crash, please post both here.
2. Once you've measured something, but not confirmed that measurement, what do you (the user) *expect/want to happen* when you click on a label. Are you trying to move it? Delete it? Use it to edit the dimension you just measured?

semhustej, 2025-10-02:
> 
>     2. I think, when I click to label I expect to just close the label  but not the masure tool . What do you think?  [@semhustej](https://github.com/semhustej)

I agree with that clicking the label should just reset the selection (basically what it does now but without the error message). 
Currently you can drag the active measurement label.