FreeCAD/FreeCAD #19712: PartDesign Feature: It is not possible to tapper the counterbore head of the screw

leoheck, 2025-02-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

It could have an option to tapper the counterbore screw head
[image]

### Full version info

```shell
1.1-dev
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leoheck, 2025-02-19:
also, the switch direction does not seem to work.

maxwxyz, 2025-02-19:
Counterbore is not tapered like a countersink. @FreeCAD/cad-advisory-group is this a valid feature request?

leoheck, 2025-02-19:
It is not, but this would help a lot when adding draft angles everywhere for injection molding.

pierreporte, 2025-02-19:
It would indeed be interesting for injection molding. I haven’t seen such a parameter before though but it sounds legitimate. There are a lot of things on the diagram already so it would be hard to integrate properly. @FreeCAD/design-working-group

leoheck, 2025-02-19:
Honestly, the existing tapered (body) angle is not integrated. It is just under the image.

Based on that I can quickly suggest that, when counterbore is selected, a new field under it (a new second item in the form), appears with a checkbox and an input field asking if the `Tapered head` is necessary as it is in the tapered (body) below.

maxwxyz, 2025-02-22:
@alfrix FYI

luzpaz, 2025-05-06:
@leoheck can you link to the new feature you're referring to ?

leoheck, 2025-05-06:
@luzpaz I was referring to this "Include in depth" that I think I had never seen before.

[image]

leoheck, 2025-09-04:
This could be a good feature for the 1.1.

leoheck, 2025-09-04:
In short, when setting the `Heat Type` to `Counterbore` or `Counterdrill`.
A new `Tapered Head` input field could appear, allowing it to be made injection molding ready.

longrackslabs, 2025-09-10:
> @longrackslabs <https://github.com/longrackslabs> When the Screw Head is
> Counterbore (this is also the case for the Counterdrill). It adds a
> straight hole for the screw head to fit in.
>
> However, when doing injection molding, we need some draft angle.
> Most of the features have a way to specify this draft angle, eg. the Pad,
> Pocket, and others too.
>
> This is what I am requesting..
> When the Counterbore or Counterdrill is selected.
> A new field appears, or is enabled to allow tapering the hole of the screw
> head.
>
> *Note that this is not the same as using Countersink.*
>
> Let me know if this image makes it clear.
> image.png (view on web)

All right, I think I get it now.   It would be helpful if you created a new
issue as requester.

I haven't really done any drawing/rending  related changes or bug fixes So
no promises this might be beyond my capabilities, but it'll be fun to give
it a go.

longrackslabs, 2025-12-16:
> it'll be fun to give it a go.
>
> @longrackslabs <https://github.com/longrackslabs> any luck ?

Sorry - I think this is beyond my capabilities.

George