FreeCAD/FreeCAD #19714: FEM: Mesh export to med format doesn't work

FEA-eng, 2025-02-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

It seems that FEM mesh export to .med format doesn't work - nothing happens (no errors, the file just doesn't appear). The same happens in 1.0.0 and 0.21.2 so it's not a recent regression. Other FEM mesh formats work normally. I tried safe mode as well.

@marioalexis84 FYI

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40303 (Git) Conda
Build type: Release
Branch: main
Hash: 145af5cddc9774f00fd4dfb3bd35e94efc5c70ca
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/81.5973
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

marioalexis84, 2025-02-20:
I can export/import without problems.
The export is done directly by SMESH, can you test it on your Linux system?

FEA-eng, 2025-02-21:
> I can export/import without problems. The export is done directly by SMESH, can you test it on your Linux system?

Strange. It doesn't work for me on Ubuntu.

```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40303 (Git) Conda
Build type: Release
Branch: main
Hash: 145af5cddc9774f00fd4dfb3bd35e94efc5c70ca
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/95.9552
```

marioalexis84, 2025-02-23:
also doesn't work from the command line?
`yourFEMMesh.FemMesh.write("pathToFile.med")`

FEA-eng, 2025-02-24:
> also doesn't work from the command line? `yourFEMMesh.FemMesh.write("pathToFile.med")`

Yeah, it doesn't work this way either.

marioalexis84, 2025-02-27:
It seems you are using some builds that were compiled using an external version of `SMESH`, so `MED` library is not linked since it is not a requirement when using the cmake option `FREECAD_USE_EXTERNAL_SMESH=ON`: https://github.com/FreeCAD/FreeCAD/blob/0ca76cc46467fa4a92fa8a039cde9428c6573777/cMake/FreeCAD_Helpers/SetupSalomeSMESH.cmake#L81
You can check the `FREECAD_USE_EXTERNAL_SMESH` option in the CMakeCache.txt file in your build directory.

The same goes for the AppImage.

@chennes , @looooo  Is there any reason not to link MED?

FEA-eng, 2025-02-28:
> It seems you are using some builds that were compiled using an external version of `SMESH`, so `MED` library is not linked since it is not a requirement when using the cmake option `FREECAD_USE_EXTERNAL_SMESH=ON`:

> You can check the `FREECAD_USE_EXTERNAL_SMESH` option in the CMakeCache.txt file in your build directory.

I have this issue with ready-made builds - 1.0 installed using official installer from the FreeCAD website and 1.1 weekly builds (for Windows and AppImage) from GitHub.

looooo, 2025-02-28:
> @chennes , @looooo Is there any reason not to link MED?

I never had a use case for med files. Also nobody was asking for it before. I guess it shouldn't be too much trouble to enable it in the conda builds.

FEA-eng, 2026-09-11:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 26.3.0dev.20260902 (Git shallow)
Build date: 2026/09/02 02:30:22
Build type: Release
Branch: (HEAD detached at c68c45227)
Hash: c68c452275ec188f35c02f6b30aba880b3783cf1
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
OpenGL version/vendor/renderer: 4.6.0 Compatibility Profile Context 26.5.2.260413/ATI Technologies Inc./AMD Radeon(TM) Graphics
```