FreeCAD/FreeCAD #19715: Draft: No preference to disable Make Face by default

FEA-eng, 2025-02-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The preference for the `Make Face` property was removed in 1.0.0 because apparently it wasn't working as expected. So currently there's no preference for this and `Make Face` is always enabled meaning that a face is created for each closed contour drawn in Draft. I would like to request a new preference that could change this behavior and keep `Make Face` disabled by default.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40303 (Git) Conda
Build type: Release
Branch: main
Hash: 145af5cddc9774f00fd4dfb3bd35e94efc5c70ca
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/81.5973
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-02-19:
This would be useful for the BIM workbench as well. Currently, whenever you want to create walls from a closed contour, you need to first disable Make face every single time.

Roy-043, 2025-02-19:
IMO there is a misunderstanding (or a lack of knowledge) here. There is still a parameter, but like other command modes it stores the last setting used, in this case the `Filled` checkbox.

Roy-043, 2025-02-19:
I am going to be blunt... Closing...

FEA-eng, 2025-02-19:
Let’s document it then.

FEA-eng, 2025-02-19:
However, a feature request to add a preference instead of the current behavior depending on the last "Filled" checkbox setting could still make sense. It's less important though.

yorikvanhavre, 2025-02-20:
Maybe we need to change/disambiguate this "Filled" checkbox (it comes from prehistoric Draft where it did only change between Wireframe/FlatLines display modes) and replace it with "Make Face"?

furgo16, 2025-02-20:
I thik that would be a good compromise. It would make the mental association more obvious.

For those of us who just learnt about this, I think we're not alone: I've seen plenty of YouTube BIM tutorials where the authors every time first create a draft object, then go to the Property Editor to set `Make Face` to `false` and finally create the wall.

PR #19738 (2025-02-24): Draft: change 'Filled' to 'Make face' in UI
Fixes #19715.

The related parameter has been renamed as well and other references to 'fill mode' have been updated.
commit 074807c3 (2025-02-24): Draft: change 'Filled' to 'Make face' in UI

Fixes #19715.

The related parameter has been renamed as well and other references to 'fill mode' have been updated.
closed 2025-02-24 by chennes