FreeCAD/FreeCAD #19721: BIM: Some walls not drawn

psommerfeld, 2025-02-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In attached project, certain Walls from "int" sketch are not drawn. In the sketch you can see lines that have now BIM Walls.
I tried to make a simpler project to reduce the complexity of my drawing but wasn't successful.

[image]

[image]

[bim.zip](https://github.com/user-attachments/files/18876421/bim.zip)

### Full version info

```shell
OS: macOS 14.0
Architecture: arm64
Version: 1.1.0dev.40176 (Git) Conda
Build type: Release
Branch: main
Hash: 2745f436026b6f1fb84dbaaf89ea22b5fb4c2295
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 72/127.5
Installed mods: 
  * fasteners.backup1686222008.1392 0.3.50 (Disabled)
  * ProDarkThemePreferencePack 1.0.0
  * BIM021 2021.12.0
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

paullee0, 2025-02-20:
This happens when you are adjacent edge segment too close, e.g. 4", while the width of wall is less than that, find those edges and adjust e.g. to 6" and it would construct.

[image]
[image]

[image]
[image]

[image]
[image]

paullee0, 2025-02-20:
[bim2_r.zip](https://github.com/user-attachments/files/18893411/bim2_r.zip)

psommerfeld, 2025-02-28:
Shouldn't FC at least display a message like "Some walls not drawn because line segments are shorter than wall width"?
This would go a long way towards useability.
It's obvious now why the walls weren't drawn, but as a new user, I initially had no idea.

paullee0, 2025-03-01:
Agreed.  As discussed in the [FC forum thread](https://forum.freecad.org/viewtopic.php?p=813062#p813062), there are 2 issues here to be precise.

1. First is when the edge which are shorter than wall width, but this alone is not necessarily problematic.  It creates problem when the edges are at an angle which make the consecutive edges self-intersecting.  This would not happens as it does not make sense on plan mostly.  Anyway, needs to find a way to detect this and report it, on the Reporting View?

2. Another is probably tolerance problem in Part.getSortedClusters(skGeomEdges) that the resulted edges cluster can't form wire.

Does it make sense with this message - "Some walls not drawn because line segments are self-intersecting" ?

paullee0, 2025-03-11:
It makes sense with wall not just perpendicular to each other wall.  It just went wrong when a segment is 'too' short.

paullee0, 2025-03-15:
Similar case reported by [psommerfeld](https://github.com/psommerfeld) on FreeCAD forum

https://forum.freecad.org/viewtopic.php?p=816758#p816758

furgo16, 2025-03-17:
Would it make sense to reopen this issue, or is it a duplicate? I can't figure out the reason it was closed for.

paullee0, 2025-03-17:
See study / potential bug fix in https://forum.freecad.org/viewtopic.php?p=817034#p817034

paullee0, 2025-03-22:
OK, anyone would like to test?

No more error and broken object, return warning.

[faces.py.TXT](https://github.com/user-attachments/files/19399179/faces.py.TXT)

[image]

paullee0, 2025-03-23:
PR submitted
- https://github.com/FreeCAD/FreeCAD/pull/20395

paullee0, 2025-04-04:
> PR submitted
> 
>     * [[Draft-Faces] Bind: Improve fuse and warning #20395](https://github.com/FreeCAD/FreeCAD/pull/20395)

This is merged, @psommerfeld can close this issue?

Thanks.

furgo16, 2025-04-04:
@paullee0, just a quick FYI for the future in case it helps: you can add a line to a PR description with the keyword `Fixes: #issuenumber` and the issue will be linked to it and auto-closed when the PR is merged.

paullee0, 2025-04-04:
I see, thanks for the heads-up.

paullee0, 2025-05-01:
Closing this now

FEA-eng, 2025-11-08:
Recomputing adds some walls, but at least one seems to be skipped.

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.20251104 (Git shallow)
Build date: 2025/11/04 19:07:44
Build type: Release
Branch: main
Hash: af014a1bb44264d749ba9522917f0d1ac76c4bbf
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
```

paullee0, 2025-11-08:
The PR make ArchWall return a warning message that a portion of edges are self-intersecting when user adds edges in a Sketch.

User needs to fix the edges in the Sketch though.

https://github.com/FreeCAD/FreeCAD/issues/19721#issuecomment-2779582417

paullee0, 2026-02-17:
@psommerfeld Should this be closed?  Thanks.

closed 2026-02-17 by psommerfeld