FreeCAD/FreeCAD #19722: Wall alignment suddenly changes

psommerfeld, 2025-02-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In attached image, you can see BIM Wall (yellow) aligned left of underlying sketch (red) when looking beginning from the bottom-left corner of the image.
As you follow the wall up, it is correct until after the second opening, where. after a short piece of wall, it suddenly changes to be aligned to the right of the underlying sketch.

[image]

[bim2.zip](https://github.com/user-attachments/files/18876597/bim2.zip)

### Full version info

```shell
OS: macOS 14.0
Architecture: arm64
Version: 1.1.0dev.40176 (Git) Conda
Build type: Release
Branch: main
Hash: 2745f436026b6f1fb84dbaaf89ea22b5fb4c2295
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 72/127.5
Installed mods: 
  * fasteners.backup1686222008.1392 0.3.50 (Disabled)
  * ProDarkThemePreferencePack 1.0.0
  * BIM021 2021.12.0
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-02-20:
I did not find the exact position in your example file, but I think this is the most probable culprit:
The Wall object in the example has `Align` property set to `Left`. 

This means the wall is aligned to the left of the sketch lines. The left direction is gathered from each of the sketch line orientation and the orientation is based on the first and the second point (the order they were created).

There are two options to solve this:

1. Use only center alignment
2. Adjust orientation of lines, which have unwanted orientation

This is a known issue: https://github.com/FreeCAD/FreeCAD/issues/11957

paullee0, 2025-02-20:
As Semhustej points out, it is how Sketch currently works with ArchWall.

closed 2025-02-21 by maxwxyz

semhustej, 2025-02-21:
@paullee0 

1. is there a faster way to switch the direction of line inside of sketch object than to delete it and recreate it?