FreeCAD/FreeCAD #19723: SVG: transform attribute is not handled collectly.

totake0224, 2025-02-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I found transform attribute is not handled collectly.
for example.
1, transform= translate(10,20) scale(2)
2, transform= scale(2) translate(10,20) 
is not same. first one is move to 10,20 and scaling.
second one is first scaling ,so translate parameter is scaled. (move to (10,20)×2)
but FreeCAD import handled it reversed order.
No1 sample
```
<svg xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" style="width: 300px;" viewBox="0 0 200 60">
    <circle id="myCircle1" cx="0" cy="0" r="10" fill="blue" transform="translate(10 20) scale(2)"/>
    <circle id="myCircle2" cx="0" cy="0" r="10" fill="red" transform="scale(2)"/>
</svg>
```
loaded at inkscape
[image]
imported at FreeCad
[image]

No2 sample
```
<svg xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" style="width: 300px;" viewBox="0 0 200 60">
    <circle id="myCircle1" cx="0" cy="0" r="10" fill="blue" transform="scale(2) translate(10 20) "/>
    <circle id="myCircle2" cx="0" cy="0" r="10" fill="red" transform="scale(2)"/>
</svg>
```
loaded at inkscape
[image]
imported at FreeCad
[image]

I think I could fix this, I will make PR soon.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39985 (Git)
Build type: Release
Branch: main
Hash: 68850593814c8dbd1bb5437034e7140de65260ca
Python 3.12.7, Qt 6.8.1, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: Japanese/Japan (ja_JP)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical/physical DPI: 96/141.585
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-22:
are they imported correctly on a TechDraw page?

totake0224, 2025-02-25:
I had confirmed this at Part Design page. so I confirmed at TechDraw page. but result is same.

PR #19845 (2025-03-02): fix #19723 SVG: transform attribute is not handled collectly
Fix #19723.
Reverse order of matrix multiply.
I intend to include one file(importSVG) . if anything else included. it my failure of making pull request.
sorry.
commit 9aef91b3 (2025-03-02): fix #19723 SVG: transform attribute is not handled collectly (#19845)

* * Move pyopen with encoding to utils.
 and modify all importing library to use pyopen with encoding.
with this change, DXF OCA AirfoilDAT with multibyte sequence always read as utf-8.

* Fix SVG transform attribute is not handler collectly.
closed 2025-03-02 by chennes