FreeCAD/FreeCAD #19732: FreeCad "Report an Issue" improvement

leoheck, 2025-02-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

We have this `Freecad > Help > Report an Issue` menu, that launches the browser on the issues page. 

This option could be renamed to `Check existing issues`.
And this `Report an Issue` could open the `New Issue` page with the `Full version info` filled already.

Thoughts?

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40303 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 145af5cddc9774f00fd4dfb3bd35e94efc5c70ca
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell /.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/145.143
Installed mods: 
  * Assembly4 0.50.18-dev
  * sheetmetal 0.7.5
  * Curves 0.6.54
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * GDT
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.26
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-20:
No, because people already create bad issues and do not search for existing ones.

closed 2025-02-20 by maxwxyz

leoheck, 2025-02-20:
If people already create bad issues, how other people can be successful searching for them? If the issues are bad, not meaningful, lacking information, poorly written how the search can go well? This is the managers task, to ask questions and to investigate. 

Also, generally speaking issue reporters are users that want to help the platform, they are not necessarily developers, or good developers. They are also not finding bugs, they are actually working trying to transform an idea into a design that most of the time is a hard and complex task when the tool helps, and when the tool doesn't help they are struggling to move on.
 
The platform could help them instead of putting walls.