FreeCAD/FreeCAD #19740: BIM: Layer created in the BIM_Layers window have 'Hide Children' property set to 'true'

semhustej, 2025-02-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

**Feature**

Layers created in the BIM_Layers window have `Hide Children` property set to `true` . This is confusing for users and unlike any other method of creating layers. Even when layer is created in Draft_Layers dialogue do not behave this way (they have no 'Hide Children' property).

**Possible solution:**
Layers created in the BIM_Layers window should have `Hide Children` property set to `false`

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40303 (Git) Conda
Build type: Release
Branch: main
Hash: 145af5cddc9774f00fd4dfb3bd35e94efc5c70ca
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073
```

### Subproject(s) affected?

BIM

### Anything else?

The topic was discussed here: https://forum.freecad.org/viewtopic.php?p=790734#p790734

As per Yorik's statement: 
" ..... Unless the layer has a "Hide Children" property, and that property is set to True. This basically only happens for IFC layers, because in the IFC file tree we don't want the objects to appear under the layer ...."

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

yorikvanhavre, 2025-03-20:
The general idea here is that when opening an IFC file where, for example, your main Site object is part of a layer, the layer will "swallow" the site in the tree, and, although it is there ni the 3D view, nothing appears under the project root, and you have the impression that something is wrong

Indeed the way it is solved now is definitely not pretty. But I had no better idea so far...

semhustej, 2025-03-20:
> The general idea here is that when opening an IFC file where, for example, your main Site object is part of a layer, the layer will "swallow" the site in the tree, and, although it is there ni the [...]

Wouldn't it make sense then to create the layer with `HideChildren = False` and only switch to `HideChildren = True` when the layer is added to IFC project?

Bascially by default FreeCAD layers `HideChildren = False` and IFC layers `HideChildren = True`.

[image]