FreeCAD/FreeCAD #19742: TechDraw: Unable to "Insert Page using Template" on Flatpak

DaWeidmann, 2025-02-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When clicking the "Insert page using Template" button, the file picker opens with the following error message:
[image]

After that, I am able to search for the templates which are located at "/var/lib/flatpak/app/org.freecad.FreeCAD/current/active/files/share/Mod/TechDraw/Templates". Selecting a template from there works. However, I would prefer to instantly be directed to that folder, without the error message.

Also, using the "Insert Default Page" button results in an empty page being added, see the following screenshot:
[image]

To me, these issues seem to be related.

### Full version info

```shell
OS: KDE Flatpak runtime (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

WandererFan, 2025-02-22:
A blank page **is** the default template.   You can change it in the General tab of the TechDraw preferences.

You can also change the default template directory in the preferences.

I don't use flatpack, but we have had similar problems with snap.  What do you get if you enter "App.getResourceDir()" in the Python console?

DaWeidmann, 2025-02-22:
Thanks for your reply. The result of `App.getResourceDir()` is `'/app/share/'`.

I didn't know that FreeCAD defaults to a blank page. Did this behaviour change since v0.21?

Now, I changed the default template and the default template directory in the preferences to the path mentioned in the original post and FreeCAD respects that and uses those templates. However, upon restarting the app, the settings change to some paths under `/run/user/1000/*`. The settings still work and this does not affect the output of `App.getResourceDir()`.

WandererFan, 2025-02-23:
"However, upon restarting the app, the settings change to some paths under /run/user/1000/*"

Your config is changing between runs?  I wonder if the flatpak version saves to a sandboxed directory instead of ~/.config/FreeCAD?

WandererFan, 2025-02-23:
Do your preferences point to a template directory outside the flatpak environment (ex ~/FCTemplates)?

DaWeidmann, 2025-02-24:
To answer your first question: The flatpak does not seem to look in `~/.config/FreeCAD` for the config. I had some files there that were last edited in November. I deleted those files and the FreeCAD config did not change.

Sorry what do you mean by "outside the flatpak environment"? I do not have knowledge about the details of Flatpaks unfortunately...

I have set the following preferences:
'Default Template' = `/var/lib/flatpak/app/org.freecad.FreeCAD/current/active/files/share/Mod/TechDraw/Templates/A4_Landscape_TD.svg`
'Template Directory' = `/var/lib/flatpak/app/org.freecad.FreeCAD/current/active/files/share/Mod/TechDraw/Templates`

Upon restarting FreeCAD, those values have changed to
'Default Template' = `/run/user/1000/doc/f35aaa1e/A4_Landscape_TD.svg`
'Template Directory' = `/run/user/1000/doc/ceb97686/Templates`

Checking for the default template file in the terminal leads to the following:
Running `ls -Lla` in `/run/user/1000/doc/f35aaa1e` give the an error
```
ls: cannot access 'A4_Landscape_TD.svg': No such file or directory
total 0
drwx------. 2 dw dw 0  1. Jan 1970  .
dr-x------. 2 dw dw 0  1. Jan 1970  ..
-?????????? ? ?  ?  ?             ? A4_Landscape_TD.svg
```

As it seems this in a problem with the Flatpak version from Flathub, would it help if i post this issue in [https://github.com/flathub/org.freecad.FreeCAD](https://github.com/flathub/org.freecad.FreeCAD)?

WandererFan, 2025-02-24:
I'm not any sort of flatpak expert either, but I don't think this is a problem with flatpak itself.

By "outside the flatpak environment", I meant saving your templates to something like "~/myFCTemplates" instead of to the flatpak controlled directories like "/var/lib/flatpak/app/...".

@adrianinsaval - does flatpak change file references?

Syres916, 2025-02-24:
@WandererFan the following may help https://discourse.flathub.org/t/why-flatpak-copies-files-to-run-user-1000-doc/6596

DaWeidmann, 2025-02-24:
That makes sense. Also I found the config files in `~/.var/app/org.freecad.FreeCAD/config/FreeCAD` and have previously made a mistake about FreeCAD changing the preferences between runs.
The behaviour is as follows: If I open the TechDraw preferences and change e.g. the `Template Directory` to `/var/lib/flatpak/app/org.freecad.FreeCAD/current/active/files/share/Mod/TechDraw/Templates` (`current` and `active` seem to be symlinks as well), the setting immediately changes to a path in `/run/user/1000`. This value also gets written to the `user.cfg` in the directory mentioned above.

DaWeidmann, 2025-02-24:
However, with the default settings (Template Directory field empty), I still get the error mentioned in the original description.

What would be the expected value for `App.getResourceDir()` be on a regular install (not Flatpak)?

WandererFan, 2025-02-25:
"What would be the expected value for App.getResourceDir() be on a regular install (not Flatpak)?"

local build:
>>> App.getResourceDir()
'/home/alg/Target/FCDevl/share/'
>>> 

AppImage:
>>> App.getResourceDir()
'/tmp/.mount_FreeCACgAFHP/usr/share/'
>>>

WandererFan, 2025-02-25:
"If I open the TechDraw preferences and change..."

That is truly odd behaviour.  I just tried this with an AppImage (another sandbox approach) and it doesn't change the preference.  

I also tried using a symlink to a template directory and that worked too.

DaWeidmann, 2025-02-25:
So this issue got even weirder and it definitely has to do something with my config. 

After completely reinstalling the Flatpak (also backing-up and removing the config directory), everything works. This includes the "Insert Page using Template" button as well as the preferences. What is different now is that FreeCAD does not use the GNOME file picker anymore but instead uses a KDE/Qt File Dialog to select the template.

After restoring the previous config, the GNOME file picker was used again and the bug reappeared.

I uploaded my config files causing the bug in a public repo on my account. The setting that causes the bug is on `user.cfg` on line 913: Changing the value of `DontUseNativeDialog` to `1` causes FreeCAD to use the Qt File Dialog which opens in the correct folder.

Does this info help in any way?

DaWeidmann, 2025-02-25:
Ok, after some more research of `QFileDialog` and the use in FreeCAD/TechDraw, I found the following:

[https://docs.flatpak.org/de/latest/portals.html](https://docs.flatpak.org/de/latest/portals.html)
Some background: [https://wiki.archlinux.org/title/XDG_Desktop_Portal](https://docs.flatpak.org/de/latest/portals.html)

And also interesting: [How to get working Qt file chooser](https://github.com/flathub/flathub/issues/1410) as well as the official 
[Portal.FileChooser API Reference](https://docs.flatpak.org/en/latest/portal-api-reference.html#gdbus-org.freedesktop.portal.FileChooser)

And looking at the implementation of `Gui::FileDialog::getOpenFileName()`, I think the following is happening:

- When `DontUseNativeDialog` is enabled, `QFileDialog::getOpenFileName` is used, which opens the native GNOME dialog and uses portals. However, those portals are not correctly resolved, neither for the input directory (from `App.getResourceDir()`), nor the output directory, which "will point into the document portal fuse filesystem in /run/user/$UID/doc/" according to the API Reference above. This also explains why the settings change to something in `/run/user/1000/doc`
- When the settings is disabled, a standard widget based file dialog without portals is used. This leads to all directories being sandboxed (`/app/*`), but inside the sandbox, everything fits and the settings work.

Does this seem plausible to you?

If so, I would be glad to help in implementing a fix :)

DaWeidmann, 2025-02-25:
Aaaaand one more note: I also tried opening a File Dialog from a different workbench (Spreadsheet), which resulted in a similaar error. Should I change to title accordingly?

WandererFan, 2025-02-26:
If it happens in other WB, we should change the title.   Maybe "Gui: Flatpak does not find files"?

I think you are on the right track in your analysis.

DaWeidmann, 2025-02-26:
This seems to be similar to #12867

DaWeidmann, 2025-02-27:
Also related: [https://forum.freecad.org/viewtopic.php?t=80893](https://forum.freecad.org/viewtopic.php?t=80893)

FEA-eng, 2026-01-13:
I assume it's still relevant as confirmed by @herrdeh, but maybe someone with Flatpak could recheck just in case.

FEA-eng, 2026-11-01:
I assume that it's still relevant.