FreeCAD/FreeCAD #19744: Hole Counterbore, unable to reduce Head Diameter if Standard selected before

mgth, 2025-02-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When creating a hole. If I select standard : Iso metric / M3 and then select "Counterbore" I'm unable to reduce "Head Diameter" under 6.10mm
If I click Ok and reopen then it's possible to change value down or value can be changed in properties.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40303 (Git) Conda
Build type: Release
Branch: main
Hash: 145af5cddc9774f00fd4dfb3bd35e94efc5c70ca
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/79.7968
Installed mods: 
  * CurvesWB 0.6.12
  * Defeaturing 1.2.3
  * ExtremeProDark 2.6.4
  * fasteners 0.5.37
  * FreeCAD-Ribbon 1.6.2.6
  * FreeCAD-themes 2025.1.7
  * freecad-xr-workbench 0.9.4
  * freecad_streamdeck_addon 0.1.7
  * OpenDark 2024.3.13
  * OpenTheme 2024.9.1
  * parts_library
  * parts_library.bak
  * ProDarkThemePreferencePack 1.0.0
  * SearchBar 1.3.3
  * workfeature
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

pierreporte, 2025-02-20:
I have encountered this bug in the last months.

maxwxyz, 2025-02-22:
@alfrix FYI

alfrix, 2025-02-22:
i just confirmed, this issue is present on 0.20 meaning it's probably there since the beginning, i would have to inspect the code to see what the original author tried to achieve

karliss, 2025-03-15:
Did some debugging. Seems like the M3 was a red herring, the limit is 6.1mm no matter what metric size you choose.

The actual problem seems to be that minimum value update for the counter bore spinbox doesn't get properly updated.
There are currently 2 places changing it:
* in the constructor of TaskHoleParameters -> since the default size is 6mm, min value gets set to 6.1
* in reaction to threadDiameterChange

But changing iso metric standard value doesn't trigger the handler for threadDiameter change and thus doesn't update minimum value. I am guessing there is some code which intentionally suppresses change events caused by programmatic updates to input field.

PR #20217 (2025-03-24): PartDesign: Update counterbore min size in reaction to diameter changes
Can't rely purely on `TaskHoleParameters::threadDiameterChanged` to update it since the signal is intentionally blocked while syncing state to GUI.

Closes #19744
commit f0fe00e1 (2025-03-24): PartDesign: Update counterbore min size in reaction to diameter changes (#20217)

Can't rely purely on `TaskHoleParameters::threadDiameterChanged` to update it since the signal is intentionally blocked while syncing state to GUI.

Closes #19744
closed 2025-03-24 by kadet1090