FreeCAD/FreeCAD #19745: Segmentation fault in assembly when setting "Advanced offset" for slide joint

ii8, 2025-02-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Joining two bodies using the slide joint in the assembly workbench allows the user to configure offsets, this is done in a separate dialog, where values for translation and center can be entered. When I press "ok" after having entered the values, FreeCAD crashes.

<details>
<summary>Backtrace</summary>

```
#0  0x00007fffc7e197ec in Extrema_GenExtCS::GlobMinCQuadric (this=this@entry=0x7fffc5a8ad30, theC=..., theNbParticles=theNbParticles@entry=48, theTUVinf=..., theTUVsup=..., theTUV=...) at ../src/Extrema/Extrema_GenExtCS.cxx:684
#1  0x00007fffc7e1b1a3 in Extrema_GenExtCS::Perform (this=this@entry=0x7fffc5a8ad30, C=..., NbT=NbT@entry=12, tmin=tmin@entry=4.9999998999999686, tsup=tsup@entry=5.0000001000000109, Tol1=Tol1@entry=9.9999999999999986e-10) at ../src/Extrema/Extrema_GenExtCS.cxx:348
#2  0x00007fffc7e1bc69 in Extrema_GenExtCS::Extrema_GenExtCS (this=this@entry=0x7fffc5a8ad30, C=..., S=..., NbT=NbT@entry=12, NbU=10, NbV=10, tmin=tmin@entry=4.9999998999999686, tsup=tsup@entry=5.0000001000000109, Umin=Umin@entry=-27.006493506493513, Usup=Usup@entry=19.993506493506487, Vmin=Vmin@entry=-18.506493506493506, Vsup=11.493506493506494, Tol1=9.9999999999999986e-10, Tol2=9.9999999999999986e-10) at ../src/Extrema/Extrema_GenExtCS.cxx:168
#3  0x00007fffc7df48a8 in Extrema_ExtCS::Perform (this=this@entry=0x7fffc5a8b570, C=..., Uinf=<optimized out>, Usup=Usup@entry=5.0000481234319221) at ../src/Extrema/Extrema_ExtCS.cxx:211
#4  0x00007fffc7df4d69 in Extrema_ExtCS::Extrema_ExtCS (this=this@entry=0x7fffc5a8b570, C=..., S=..., TolC=TolC@entry=9.9999999999999986e-10, TolS=TolS@entry=9.9999999999999986e-10) at ../src/Extrema/Extrema_ExtCS.cxx:60
#5  0x00007fffca0f9e06 in IntTools_EdgeFace::CheckTouch (this=this@entry=0x55555e6472a8, aCP=..., aTx=@0x7fffc5a8b6f8: 78.900000000000006) at ../src/IntTools/IntTools_EdgeFace.cxx:380
#6  0x00007fffca0fa6ec in IntTools_EdgeFace::MakeType (aCommonPrt=..., this=0x55555e6472a8) at ../src/IntTools/IntTools_EdgeFace.cxx:332
#7  IntTools_EdgeFace::MakeType (this=0x55555e6472a8, aCommonPrt=...) at ../src/IntTools/IntTools_EdgeFace.cxx:292
#8  0x00007fffca0fb536 in IntTools_EdgeFace::Perform (this=this@entry=0x55555e6472a8) at ../src/IntTools/IntTools_EdgeFace.cxx:565
#9  0x00007fffca1d6ead in BOPAlgo_EdgeFace::Perform (this=0x55555e647258) at ../src/BOPAlgo/BOPAlgo_PaveFiller_5.cxx:136
#10 0x00007fffca1d5c09 in BOPTools_Parallel::ContextFunctor2<NCollection_Vector<BOPAlgo_EdgeFace>, IntTools_Context>::operator() (theIndex=<optimized out>, theThreadIndex=0, this=<optimized out>) at /builddir/occt-7.7.1/src/BOPTools/BOPTools_Parallel.hxx:151
#11 OSD_ThreadPool::Job<BOPTools_Parallel::ContextFunctor2<NCollection_Vector<BOPAlgo_EdgeFace>, IntTools_Context> >::Perform (this=0x7fffffff8190, theThreadIndex=0) at /builddir/occt-7.7.1/src/OSD/OSD_ThreadPool.hxx:299
#12 0x00007fffc76b1d66 in OSD_ThreadPool::performJob (theFailure=..., theJob=0x7fffffff8190, theThreadIndex=0) at ../src/OSD/OSD_ThreadPool.cxx:286
#13 0x00007fffc76b217c in OSD_ThreadPool::EnumeratedThread::performThread (this=0x55555b61b628) at ../src/OSD/OSD_ThreadPool.cxx:326
#14 0x00007fffc76b21c9 in OSD_ThreadPool::EnumeratedThread::runThread (theTask=<optimized out>) at ../src/OSD/OSD_ThreadPool.cxx:340
#15 0x00007ffff7fb858e in start (p=0x7fffc5a8c630) at src/thread/pthread_create.c:195
#16 0x00007ffff7fc552b in __clone () at src/thread/x86_64/clone.s:22
Backtrace stopped: frame did not save the PC
```

</details>

### Full version info

```shell
OS: Void Linux (wayland)
Architecture: x86_64
Version: 1.1.0dev.40263 (Git)
Build type: Release
Branch: main
Hash: eb886449c2d63b1b5e61bf6a5b7acac61d2382e5
Python 3.13.2, Qt 6.8.1, Coin 4.0.3, Vtk 9.3.1, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical/physical DPI: 96/91.7275
Installed mods: 
  * freecad.gears 1.2.0
  * fasteners 0.5.21
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ii8, 2025-02-22:
This crash also happens when using the transform tool and then cancelling before applying any transformation, I think it's the same bug because the backtrace looks very similar:

```
#0  0x00007fffc7c197ec in Extrema_GenExtCS::GlobMinCQuadric (this=this@entry=0x7fffc983ad30, theC=..., theNbParticles=theNbParticles@entry=48, theTUVinf=..., theTUVsup=..., theTUV=...) at ../src/Extrema/Extrema_GenExtCS.cxx:684
#1  0x00007fffc7c1b1a3 in Extrema_GenExtCS::Perform (this=this@entry=0x7fffc983ad30, C=..., NbT=NbT@entry=12, tmin=tmin@entry=4.9999998999999686, tsup=tsup@entry=5.0000001000000109, Tol1=Tol1@entry=9.9999999999999986e-10) at ../src/Extrema/Extrema_GenExtCS.cxx:348
#2  0x00007fffc7c1bc69 in Extrema_GenExtCS::Extrema_GenExtCS (this=this@entry=0x7fffc983ad30, C=..., S=..., NbT=NbT@entry=12, NbU=10, NbV=10, tmin=tmin@entry=4.9999998999999686, tsup=tsup@entry=5.0000001000000109, Umin=Umin@entry=-27.006493506493513, Usup=Usup@entry=19.993506493506487, Vmin=Vmin@entry=-18.506493506493506, Vsup=11.493506493506494, Tol1=9.9999999999999986e-10, Tol2=9.9999999999999986e-10) at ../src/Extrema/Extrema_GenExtCS.cxx:168
#3  0x00007fffc7bf48a8 in Extrema_ExtCS::Perform (this=this@entry=0x7fffc983b570, C=..., Uinf=<optimized out>, Usup=Usup@entry=5.0000481234319221) at ../src/Extrema/Extrema_ExtCS.cxx:211
#4  0x00007fffc7bf4d69 in Extrema_ExtCS::Extrema_ExtCS (this=this@entry=0x7fffc983b570, C=..., S=..., TolC=TolC@entry=9.9999999999999986e-10, TolS=TolS@entry=9.9999999999999986e-10) at ../src/Extrema/Extrema_ExtCS.cxx:60
#5  0x00007fffc9ef9e06 in IntTools_EdgeFace::CheckTouch (this=this@entry=0x55555d6ed2e8, aCP=..., aTx=@0x7fffc983b6f8: 78.900000000000006) at ../src/IntTools/IntTools_EdgeFace.cxx:380
#6  0x00007fffc9efa6ec in IntTools_EdgeFace::MakeType (aCommonPrt=..., this=0x55555d6ed2e8) at ../src/IntTools/IntTools_EdgeFace.cxx:332
#7  IntTools_EdgeFace::MakeType (this=0x55555d6ed2e8, aCommonPrt=...) at ../src/IntTools/IntTools_EdgeFace.cxx:292
#8  0x00007fffc9efb536 in IntTools_EdgeFace::Perform (this=this@entry=0x55555d6ed2e8) at ../src/IntTools/IntTools_EdgeFace.cxx:565
#9  0x00007fffc9fd6ead in BOPAlgo_EdgeFace::Perform (this=0x55555d6ed298) at ../src/BOPAlgo/BOPAlgo_PaveFiller_5.cxx:136
#10 0x00007fffc9fd5c09 in BOPTools_Parallel::ContextFunctor2<NCollection_Vector<BOPAlgo_EdgeFace>, IntTools_Context>::operator() (theIndex=<optimized out>, theThreadIndex=2, this=<optimized out>) at /builddir/occt-7.7.1/src/BOPTools/BOPTools_Parallel.hxx:151
#11 OSD_ThreadPool::Job<BOPTools_Parallel::ContextFunctor2<NCollection_Vector<BOPAlgo_EdgeFace>, IntTools_Context> >::Perform (this=0x7fffffff9ba0, theThreadIndex=2) at /builddir/occt-7.7.1/src/OSD/OSD_ThreadPool.hxx:299
#12 0x00007fffc74b1d66 in OSD_ThreadPool::performJob (theFailure=..., theJob=0x7fffffff9ba0, theThreadIndex=2) at ../src/OSD/OSD_ThreadPool.cxx:286
#13 0x00007fffc74b217c in OSD_ThreadPool::EnumeratedThread::performThread (this=0x55555aeaa458) at ../src/OSD/OSD_ThreadPool.cxx:326
#14 0x00007fffc74b21c9 in OSD_ThreadPool::EnumeratedThread::runThread (theTask=<optimized out>) at ../src/OSD/OSD_ThreadPool.cxx:340
#15 0x00007ffff7fb858e in start (p=0x7fffc983c630) at src/thread/pthread_create.c:195
#16 0x00007ffff7fc552b in __clone () at src/thread/x86_64/clone.s:22
Backtrace stopped: frame did not save the PC
```

ii8, 2025-02-22:
Same as the full version info in the OP

maxwxyz, 2025-02-23:
@kadet1090 FYI

kadet1090, 2025-02-23:
The backtrace indicates that error is in OCC, it should be probably reported back to them with simple reproduction scenario.

ii8, 2025-02-23:
I have upgraded from OCCT 7.7.1 to 7.9.0. FreeCAD now doesn't crash anymore.
With OCCT 7.7.1 there were a lot of crashes, FreeCAD was completely unusable, so far no crash with OCCT 7.9.0.

pieterhijma, 2025-03-06:
Since a newer version of OCCT solves this issue, shall we close it?

maxwxyz, 2025-03-06:
@pieterhijma do we release all builds with the newer version?

pieterhijma, 2025-03-07:
That's a very good question. I'm looking into this. According to @hyarion, the lowest Linux version we support is OCCT 7.5.0 (Ubuntu LTS 22.04). This will soon move to only support for Ubuntu LTS 24.04 which means that the lowest version we support is OCCT 7.6.0.

Since this issue was reported on 7.7.1, it is likely (or at least possible) that the bug exists in 7.6.0 as well. In that sense, we can't close the issue. On the other hand, it does not make much sense to report this to OCCT because they will simply say: yes fixed in a later version, so please use that version. In that sense we can close the issue. 

I suggest we leave it open then but I mark this as Done in my OCCT Liaison project, because this is not something I can report to OCCT.

hyarion, 2025-03-07:
Even if it is fixed upstream we need to support the versions that we specify.

We should either find a workaround for this issue or drop the versions where this hasn't been fixed.

oursland, 2025-03-07:
Can the issue be reproduced with `occt` 7.8.1, which is used in the weekly builds, the tagged releases, the freecad-daily PPA, and freecad-flatpak? Or 7.8.0-blobfish which is used on freecad-snap?

pieterhijma, 2025-03-13:
I can actually not even reproduce on OCCT 7.7.1:

```
OS: Arch Linux (xcb)
Architecture: x86_64
Version: 1.1.0dev.40609 (Git)
Build date: 2025/03/09 13:41:26
Build type: Release
Branch: main
Hash: 844cceb04f0a82f55dee46445c4d57a7e20111ef
Python 3.13.2, Qt 6.8.2, Coin 4.0.3, Vtk , OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1.33333
Installed mods: 
  * OpenTheme 2024.5.16
  * Ondsel-Lens 2024.10.21.01
  * fasteners 0.5.34
  * WebTools 1.0.0
  * osh-autodoc-workbench 0.2.3
  * DynamicData 2.70.0
```

I created two bodies in Part Design, a simple sketch with a pad. I used the Assembly insert command to insert it into the assembly where I grounded one body. Then I selected a face from each body and created a slide joint. I checked the "Show advanced offsets" and opened the separate dialog. Entered values for translation and center and pressed ok. Everything works as expected.