FreeCAD/FreeCAD #19749: Start: the UI slows down noticeably with a Custom Folder shown in the Start page

furgo16, 2025-02-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

While using the [Custom folder feature](https://github.com/FreeCAD/FreeCAD/pull/19473) on a daily  basis, I've noticed a situation where it slows down the whole FreeCAD UI, requiring a restart or closing the Start page to remove the lagginess:

1. Set CustomFolder to a folder with a fair bit of different supported file formats (e.g. my home folder, with about 40 supported files)
2. Scroll down the Start page
3. At the point of reaching the "Custom folder" area, scrolling becomes noticeably laggy, stopping for a few seconds before resuming scrolling
4. This seems to affect the whole UI afterwards: clicking on a menu (e.g. File) takes seconds until the menu is expanded. Only closing the Start page or restarting FreeCAD (and not scrolling on the Start page) brings back the UI to its usual responsiveness.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40313 (Git) Snap 1307
Build type: Release
Branch: main
Hash: e1512dc0098c3ffa6aa4b42f07d7cedb99899353
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 7.1.1, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/95.1083
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * ezydxf
  * Arch
  * parts_library.backup1739558958.8474634
  * Assembly4 0.50.17
```

### Subproject(s) affected?

Addon Manager

### Anything else?

Something else I've observed: when making the FreeCAD window narrower, the "Examples" and "Recent Files" section occupy at most two rows (and then the section becomes scrollable). The "Custom Folder" section occupies 10 rows. I can't figure out what sets the maximum number of rows shown for each section. Is this calculated depending on the number of items?

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-02-21:
My custom folder probably is in the region of 30 files and my build is self compiled but I've not noticed any slowdowns with menus or scrolling. @furgo16 is your system a single monitor setup?

furgo16, 2025-02-21:
Thanks for testing. It's a double monitor setup.

Syres916, 2025-02-21:
Wel it's cheaper for you to test powering your system up with a single monitor and see if it's any faster than me buying another monitor 😄

furgo16, 2025-02-21:
Certainly :-) I'll try later on this evening. Anything in particular I should be looking for with the single-monitor setup?

maxwxyz, 2025-02-22:
Please test in safe mode

Syres916, 2025-02-24:
@furgo16 did the single monitor test prove useful?

furgo16, 2025-02-24:
Thanks for checking. I've not yet managed. I'll report back when I do.

furgo16, 2025-02-26:
> Please test in safe mode

Unfortunately, this cannot be tested in safe mode, as "CustomFolder" isn't defined there. Even when defining it via the parameter editor, the Start Page will not acknowledge it/reload, and thus the custom folder is not shown.

I've made some progress, but I still cannot pinpoint the cause:

- Single monitor test does not make a difference. Still same behaviour
- I've noticed that what causes the slow down is CPU usage: when scrolling through the custom folder, my CPU goes to 100% and it stays like that for a while.

Some thoughts:

- Perhaps not an issue not with the number of files (~30) or their sizes (max 500kb to 1Mb), but with a particular type of file
- I've always had this situation whereby when opening a file, the default filter on the file selector is set to "All supported types". It always takes a few seconds until any file is shown. I wished the default filter were set as ".FCStd" only. Or that it remembered the last filter setting. The custom folder also uses a similar filter (all files). Perhaps related?
- So far I've tested this with a snap. I wonder if xdg-desktop-portal plays a role here. Just occasionally file selection in other snaps that use it (e.g. Firefox) gets very slow. Also perhaps related?

maxwxyz, 2025-08-19:
may be fixed by https://github.com/FreeCAD/FreeCAD/pull/23186

tetektoza, 2025-08-19:
@furgo16 if you could test it with the above change furgo, I would be grateful. I think most of the problems with the lag on Start window are being caused by thumbnails being constantly loaded. This above fix should reduce it heavily. And another thing is that we still receive events on Start page, even if we are in a separate tab. This resulted in the thumbnails being constantly to the memory even though they were not displayed.