FreeCAD/FreeCAD #19750: Windows: Missing menus after wake up from sleep mode.

mgth, 2025-02-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When pc wake up from sleep mode, main menu and toolbar are missing.
I can recover them by dragging the main window to another string. That may be linked to DPI awareness as the two screens have different DPI. The main is 175% and the second is set to 200%

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40303 (Git) Conda
Build type: Release
Branch: main
Hash: 145af5cddc9774f00fd4dfb3bd35e94efc5c70ca
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/79.7968
Installed mods: 
  * CurvesWB 0.6.12
  * Defeaturing 1.2.3
  * ExtremeProDark 2.6.4
  * fasteners 0.5.37
  * FreeCAD-Ribbon 1.6.2.6
  * FreeCAD-themes 2025.1.7
  * freecad-xr-workbench 0.9.4
  * freecad_streamdeck_addon 0.1.7
  * OpenDark 2024.3.13
  * OpenTheme 2024.9.1
  * parts_library
  * parts_library.bak
  * ProDarkThemePreferencePack 1.0.0
  * SearchBar 1.3.3
  * workfeature
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

mgth, 2025-02-21:
After wake up
[image]

Normal window
[image]

Note that I'm using Ribbon, but it was the same without it.

FEA-eng, 2025-11-23:
Seems to be still relevant since a duplicate of it was created a week ago: #25385

FEA-eng, 2026-10-17:
Likely still relevant.