FreeCAD/FreeCAD #19751: [Bim/Draft] Std_ToggleVisibility does not work for an object inside an array.

kaiwas, 2025-02-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Video demonstration.

https://github.com/user-attachments/assets/06eb6a34-49ea-41c8-9700-8210ea915f11

Checked in safe mode and even tried the release version 1.0. Cleared the cache and deleted the configuration files.

Visibility toggling starts working again if I perform other manipulations on the object (for example, mirror it)

However, literally yesterday everything worked fine.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40303 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 145af5cddc9774f00fd4dfb3bd35e94efc5c70ca
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Logical/physical DPI: 96/91.7938
Installed mods: 
  * FreeCAD_SketchArch
  * PitchedRoof
  * Road 2025.2.6
  * addFC 1.4.0
  * freecad.gears 1.3.0
  * sheetmetal 0.7.11
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-02-21:
The problem only appears with the array of links

https://github.com/user-attachments/assets/43492af8-e4f3-4a9a-9a3f-279f597ec946

kaiwas, 2025-02-21:
The problem also does not appear after reopening the file.

https://github.com/user-attachments/assets/caba6382-f9b4-4836-b87e-314276b39983

But the new array created in the reopened file will have a problem.

kaiwas, 2025-11-05:
> is still relevant

 is still relevant

Roy-043, 2026-10-10:
This may have been fixed with #33289. Needs to be tested with the next weekly.