FreeCAD/FreeCAD #19754: Delete function improvement

ErtugrulEge, 2025-02-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Greetings to the Community!

Currently FC delete function dosn't delete dependencies.E.g. draw a wall and press delete on keyboard,FC only deletes the wall but not its dependency line or sketch. You have to press second time to completely delete object.This is a time consuming process. 

There are some options about deleting: 

* (Shift+Del on windows) sends wall+sketch to a trash folder 

* or you can make a  folder e.g named  recycle and move elements to that folder and after delete that folder

but, Isn't there can be  an option that deletes object and its dependencies all at once, on preferences General tab ?

I want to press delete button  only one time. 

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40239 (Git) Conda
Build type: Release
Branch: main
Hash: 3623f8baf50f924fd2a213495c72a5e316cf9341
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/81.5973
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-02-21:
This would be a great opportunity!

luzpaz, 2025-05-13:
Can someone offer a less generic title for this ticket?

leoheck, 2025-08-21:
I found a workaround. Not sure if it works all the time, but it is better than nothing. 
1. Create a new Part
2. Drag the things you want to delete inside.
3. Delete the part. Fc will ask if you want to the the contents recursively.