FreeCAD/FreeCAD #19756: Broken links crashing freecad

leoheck, 2025-02-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

This happened 3 times already.
Takes couple of minutes working to crash but it happens.

```
Fillet011: BRep_API: command not done
<Exception> ObjectIdentifier.cpp(1698): Document 'Battery_Cover' not found in 'Battery_Cover#LCS_Origin.Placement'
in property binding 'Ground_Sensor_1_3#Battery_Cover.Placement'
Battery_Cover: Document 'Battery_Cover' not found in 'Battery_Cover#LCS_Origin.Placement'
in property binding 'Ground_Sensor_1_3#Battery_Cover.Placement'
Body_Front: Link broken!
Object: Body_Front
File: Body.FCStd
<Exception> ObjectIdentifier.cpp(1698): Document 'Battery_Cover' not found in 'Battery_Cover#LCS_Origin.Placement'
in property binding 'Ground_Sensor_1_3#Battery_Cover.Placement'
Battery_Cover: Document 'Battery_Cover' not found in 'Battery_Cover#LCS_Origin.Placement'
in property binding 'Ground_Sensor_1_3#Battery_Cover.Placement'
Body_Front: Link broken!
Object: Body_Front
File: Body.FCStd
<Exception> ObjectIdentifier.cpp(1698): Document 'Battery_Cover' not found in 'Battery_Cover#LCS_Origin.Placement'
in property binding 'Ground_Sensor_1_3#Battery_Cover.Placement'
Battery_Cover: Document 'Battery_Cover' not found in 'Battery_Cover#LCS_Origin.Placement'
in property binding 'Ground_Sensor_1_3#Battery_Cover.Placement'
Body_Front: Link broken!
Object: Body_Front
File: Body.FCStd
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45330) [0x7323dce45330]
#1  0x7323919f7882 in TopLoc_Location::Multiplied(TopLoc_Location const&) const from /tmp/.mount_FreeCAhhIlAK/usr/lib/./libTKMath.so.7.7+0x12
#2  0x7323b8151ec0 in BRep_Tool::Surface(TopoDS_Face const&, TopLoc_Location&) from /tmp/.mount_FreeCAhhIlAK/usr/lib/./libTKBRep.so.7.7+0x30
#3  0x7323b8166a17 in BRepAdaptor_Surface::Initialize(TopoDS_Face const&, bool) from /tmp/.mount_FreeCAhhIlAK/usr/lib/./libTKBRep.so.7.7+0x97
#4  0x7323b8167d23 in BRepAdaptor_Surface::BRepAdaptor_Surface(TopoDS_Face const&, bool) from /tmp/.mount_FreeCAhhIlAK/usr/lib/./libTKBRep.so.7.7+0xf3
#5  0x7323cc093809 in Measure::Measurement::planePlaneDistance() const from /tmp/.mount_FreeCAhhIlAK/usr/lib/Measure.so+0xc9
#6  0x7323accc155e in MeasureGui::QuickMeasure::printResult() from /tmp/.mount_FreeCAhhIlAK/usr/lib/MeasureGui.so+0x42e
#7  0x7323accc1ded in MeasureGui::QuickMeasure::processSelection() from /tmp/.mount_FreeCAhhIlAK/usr/lib/MeasureGui.so+0x5d
#8  /tmp/.mount_FreeCAhhIlAK/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7323dd4eb6f2]
#9  0x7323dd4ef34a in QTimer::timeout(QTimer::QPrivateSignal) from /tmp/.mount_FreeCAhhIlAK/usr/bin/../lib/libQt5Core.so.5+0x2a
#10  0x7323dd4e064d in QObject::event(QEvent*) from /tmp/.mount_FreeCAhhIlAK/usr/bin/../lib/libQt5Core.so.5+0x7d
#11  0x7323de3703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAhhIlAK/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#12  0x7323e0178639 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAhhIlAK/usr/bin/../lib/libFreeCADGui.so+0xc9
#13  0x7323dd4b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAhhIlAK/usr/bin/../lib/libQt5Core.so.5+0x108
#14  0x7323dd509f51 in QTimerInfoList::activateTimers() from /tmp/.mount_FreeCAhhIlAK/usr/bin/../lib/libQt5Core.so.5+0x381
#15  /tmp/.mount_FreeCAhhIlAK/usr/bin/../lib/libQt5Core.so.5(+0x30a75c) [0x7323dd50a75c]
#16  /tmp/.mount_FreeCAhhIlAK/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x7323db710ad7]
#17  /tmp/.mount_FreeCAhhIlAK/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x7323db713ac7]
#18  /tmp/.mount_FreeCAhhIlAK/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7323db7140ce]
#19  0x7323dd50aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAhhIlAK/usr/bin/../lib/libQt5Core.so.5+0x66
#20  0x7323dd4b570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAhhIlAK/usr/bin/../lib/libQt5Core.so.5+0x11b
#21  0x7323dd4bd212 in QCoreApplication::exec() from /tmp/.mount_FreeCAhhIlAK/usr/bin/../lib/libQt5Core.so.5+0x82
#22  0x7323e00d858a in Gui::Application::runApplication() from /tmp/.mount_FreeCAhhIlAK/usr/bin/../lib/libFreeCADGui.so+0x8aa
#23  /tmp/.mount_FreeCAhhIlAK/usr/bin/freecad(+0x68f6) [0x63ced0ff58f6]
#24  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x7323dce2a1ca]
#25  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x7323dce2a28b]
#26  /tmp/.mount_FreeCAhhIlAK/usr/bin/freecad(+0x6c7e) [0x63ced0ff5c7e]

```

Cannot share the file. 
I don't have the steps to reproduce it yet since I don't have time right now to exercise this.
But I will give it a try later.

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40303 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 145af5cddc9774f00fd4dfb3bd35e94efc5c70ca
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/145.143
Installed mods: 
  * Assembly4 0.50.18-dev
  * sheetmetal 0.7.11
  * DynamicData 2.71.0
  * Curves 0.6.58
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * GDT
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.1.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.2
```

### Subproject(s) affected?

Core

### Anything else?

I am using Asm4 by the way.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ii8, 2025-02-22:
This happens to me too. Steps to reproduce: make 3 files like this:
File F1 containing spreadsheet S and body B1
File F2 containing body B2 that depends on spreadsheet S
File F3 containing assembly A that contains B1 and B2

Then ensure partial file loading is enabled and do this:

Open file F2, now F1 is partially loaded because it needs S
Open F3, now FreeCAD complains about a broken link
Try to delete the broken object B1 in the assembly
Crash

Workaround: just turn off partial loading in settings

maxwxyz, 2025-02-22:
Does this also happen in 1.0?

leoheck, 2025-02-22:
> Try to delete the broken object B1 in the assembly
Crash

For me, it crashes as soon as I click on an LCS file.

In my case, I don't have this partial loading selected. This is what I use.

[image]

leoheck, 2025-02-25:
The Partial Loagin seems to work even if the settings are disabled. Unless I am not understanding how this works.
When I open an Assembly, the linked files open too and they are not loading completely, but the settings say the partial loading is disabled.

ii8, 2025-02-25:
You can also try updating opencascade to 7.9.0, that fixed some problems for me.

leoheck, 2025-02-25:
Ah, that's right, I did not read ot carefully. It is a bit odd, but I didn't help myself either.

leoheck, 2025-02-26:
> You can also try updating opencascade to 7.9.0, that fixed some problems for me.

This should fix some problems, indeed. It could be an effort for the weekly versions.

leoheck, 2025-02-26:
> Ticking the box disables it, it's a bit odd. Tick = disabled, no tick = enabled. In your screenshot above partial loading is enabled.

This option could be `Enable partial load` coming checked by default.

Zolko-123, 2025-02-27:
> > Ticking the box disables it, it's a bit odd. Tick = disabled, no tick = enabled. In your screenshot above partial loading is enabled.
> 
> This option could be `Enable partial load` coming checked by default.

yes, that option is reversed. You should **disable** partial loading (especially as it doesn't work with Asm4 at all, and it's not very useful anyway : what it does is to load the Brep geometries from the file without loading the XML data)

stefano2734, 2026-09-12:
How is the state here? Is it reproducing in higher than 7.8.1?
Perhaps it should be closed after this history here.
Actual 7.9.3 and 8.0.1. so should it be tested with newer version 1.1.3 and 26.3.

luzpaz, 2026-09-12:
@ii8 can you upload these example files for convenient testing for testers ? (https://github.com/FreeCAD/FreeCAD/issues/19756#issuecomment-2676177470)

ii8, 2026-09-12:
It's been a long time and I don't remember what files I was working on at the time. Since then I've switched to Dune3D now as well, so unfortunately I don't think I have anything useful for you.

But if the issue I was having with OCCT 7.7 is affecting anyone they would notice immediately without any specially crafted reproducer because crashes were very frequent so that FreeCAD was almost unusable. I don't think my exact problem was reproducible by others, I suspect the issue in the OP was subtly different from mine, perhaps a quirk of my system.

luzpaz, 2026-09-12:
@leoheck have you run into this in any other context ?

stefano2734, 2026-09-22:
I would say with this unclear behavior and without files: nothing for occt liaison.

maxwxyz, 2026-09-22:
The weekly tomorrow will be with OCC8. If fixed in 7.8.1 I'll close this. If the issue is still reproducible in tomorrows weekly, just comment.

closed 2026-09-22 by maxwxyz