FreeCAD/FreeCAD #19757: pyException when importing certain excel spreadsheets

pageeddie, 2025-02-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

If an excel spreadsheet has had a sheet deleted, FreeCAD can no longer properly import it and will generate a pyException. To reproduce the problem (demonstrating correct behavior first):
1) In Excel create a new spreadsheet - by default it has a single sheet called "Sheet1"
2) in Excel, create a new sheet within the same spreadsheet. By default it will be called "Sheet2"
3) insert whatever data you want in the two sheets. Save the spreadsheet.
4) in FreeCAD, go to File > Import, and import this spreadsheet. THIS WORKS FINE. A FreeCAD spreadsheet is created for both Sheet1 and Sheet2.
5) In Excel, delete Sheet1 and save the spreadsheet again.
6) in FreeCAD, go to File > Import, and import this spreadsheet. Report view says:
            08:39:58  pyException: Traceback (most recent call last):
              File "<string>", line 2, in <module>
              File "C:\Users\eddie\Desktop\f\FreeCAD_weekly-builds-40303-conda-Windows-x86_64-py311\bin\Lib\site-packages\freecad\module_io.py", line 16, in OpenInsertObject
                getattr(importerModule, importMethod)(objectPath, *importArgs, **importKwargs)
              File "C:\Users\eddie\Desktop\f\FreeCAD_weekly-builds-40303-conda-Windows-x86_64-py311\Mod\Spreadsheet\importXLSX.py", line 470, in insert
                f = z.open("xl/worksheets/" + sheetFile)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
              File "C:\Users\eddie\Desktop\f\FreeCAD_weekly-builds-40303-conda-Windows-x86_64-py311\bin\Lib\zipfile.py", line 1564, in open
                zinfo = self.getinfo(name)
                        ^^^^^^^^^^^^^^^^^^
              File "C:\Users\eddie\Desktop\f\FreeCAD_weekly-builds-40303-conda-Windows-x86_64-py311\bin\Lib\zipfile.py", line 1493, in getinfo
                raise KeyError(
            <class 'KeyError'>: "There is no item named 'xl/worksheets/sheet2.xml' in the archive"

Note that I have managed to implement a code fix on my own... But I am new to GitHub, I know essentially nothing about the internals of excel files, and I am not super experienced with Python. Nonetheless, here is the fix I did that seems to work.
In Mod/Spreadsheet/importXLSX.py, there is a function "handleWorkBook". This has the lines:
        idRef = sheetAtts.getNamedItem("sheetId")
        sheetFile = "sheet" + getText(idRef.childNodes) + ".xml"
I changed these two lines to
        idRef = sheetAtts.getNamedItem("r:id")
        sheetFile = "sheet" + getText(idRef.childNodes)[3:] + ".xml"        
 
I hope this is helpful!

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40303 (Git) Conda
Build type: Release
Branch: main
Hash: 145af5cddc9774f00fd4dfb3bd35e94efc5c70ca
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/93.2987
Installed mods: 
  * Curves 0.6.58
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-22:
Please add an example file. Does it also behave like this in the 1.0 version?

pageeddie, 2025-02-22:
[breaks_freeCAD.xlsx](https://github.com/user-attachments/files/18924756/breaks_freeCAD.xlsx)

This does crash in 1.0 - in fact, this is where I originally observed it. But it definitely happens in 1.1.0dev as well.
The excel file was created with 3 sheets. Sheet1, Sheet2, Sheet3. In Excel, I just deleted Sheet2 and saved the result.

pageeddie, 2025-02-22:
So I tried out FreeCAD 1.0 stable release, in safe mode. The behavior is the same - that is to say, I get a pyException

luzpaz, 2025-04-08:
@pageeddie will you make a PR for the dev branch ITC? Worst case,  we won't make a backport to 1.0 (unless someone figure out how to)

godiard, 2025-05-31:
I have tested the solution provided by @pageeddie on the dev version and provided a pr  https://github.com/FreeCAD/FreeCAD/pull/21686

closed 2025-06-05 by chennes

3x380V, 2025-06-05:
@chennes, please see 347e2d2b22fa437621634c9d0ca4512c1ef0d950. A question immediately coming in mind is: if the fix proposed in the issue was correct, why would Werner bother to further complicate it?

chennes, 2025-06-05:
@3x380V I don't know -- I tested the solution in #21686 against the file here, but not against anything else. Can you create a spreadsheet that still fails?

3x380V, 2025-06-05:
@chennes, no, I can give it a try. But consider the specification Werner linked in his commit. Current solution still assumes the sheet name is `sheet<something>.xml` and if that assumption fails, so does import.

3x380V, 2025-06-05:
Modified [breaks_freeCAD.xlsx](https://github.com/user-attachments/files/20612877/breaks_freeCAD.xlsx) to still fail:
```
process Strings: 
string:  a
string:  c
Traceback (most recent call last):
  File "<string>", line 2, in <module>
  File "/home/ladis/src/FreeCAD-build-qt5/Ext/freecad/module_io.py", line 16, in OpenInsertObject
    getattr(importerModule, importMethod)(objectPath, *importArgs, **importKwargs)
  File "/home/ladis/src/FreeCAD-build-qt5/Mod/Spreadsheet/importXLSX.py", line 470, in insert
    f = z.open("xl/worksheets/" + sheetFile)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/zipfile.py", line 1553, in open
    zinfo = self.getinfo(name)
            ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/zipfile.py", line 1482, in getinfo
    raise KeyError(
<class 'KeyError'>: "There is no item named 'xl/worksheets/sheet5.xml' in the archive"
```

3x380V, 2025-06-05:
@chennes, #21785 fixes that. I added a few more fixes there, see comment.

chennes, 2025-06-05:
Yeah, I'll merge that without the test, then make a separate PR to add the test