FreeCAD/FreeCAD #19767: [Feature] Block user from entering invalid variable name

NXTler, 2025-02-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Greetings,
it's very easy for the user to enter a variable name that will prevent them from using it in an expression, like spaces or minuses. Furthermore, the error resulting message is not really clear, making learning expressions unnecessarily hard.
We should prevent, or at least warn the user about names that prevents the use of expressions.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40303 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 145af5cddc9774f00fd4dfb3bd35e94efc5c70ca
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Logical/physical DPI: 96/127.182
Installed mods: 
  * sheetmetal 0.7.5
  * A2plus 0.4.68
  * QuickMeasure 2022.10.28
  * Curves 0.6.54
  * CfdOF 1.29.6
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * kicadStepUpMod 11.5.2
```

### Subproject(s) affected?

Expressions

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

NXTler, 2025-03-02:
Oh, I forgot to write a reproduce section. The steps are easy:
1. Create a sketch with some geometry in it
2. Give the dimensions names like `test test`, `test-test`, `/test`
3. Create a second sketch
4. Try to get the constraints with `Sketch.Constraints.[name]`

NXTler, 2025-03-02:
As naming Dimensions might not be done for the use in expression, but rather for organization, a warning including the specific character should be the better solution.

NXTler, 2025-03-02:
As far as I can tell, the naming issue persists everywhere you can enter an expression (including Spreadsheet), it doesn't matter if you use `Sketch[some number].` or a label.

pitch-circle, 2025-03-06:
As above, for ziad-ashraf7, the issue is related to allowable syntax for names used in Expressions.

The rules are documented here:

https://wiki.freecad.org/Expressions#Invalid_characters_and_names

They are reasonably clear in what is permitted and disallowed. However, it is not an exhaustive list - good luck with the "and many more!" written there.

Perhaps an enhancement to error messages in general would be to have the facility to provide a (optional) link to the relevant section of the Wiki.

ziad-ashraf7, 2025-03-22:
Hi @leuriato , it seems that i'm busy nowadays. so you can start working on the issue. I hope you can solve it ASAP.
Good luck!

closed 2025-04-21 by kadet1090