FreeCAD/FreeCAD #19768: CAM: Adaptive producing tiny, random loops

axxel, 2025-02-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I have a trivial circular pocket (say 12mm diameter) that I wanted to clear with an endmill (say 5mm). I noticed a sudden jerking/pause on my machine. A closer look at the path revealed 2 tiny loops, where the movement is practically reversed. See screenshots:

[image]

[image]

Sometimes I can 'fix' it by playing around with the `Tolerance` setting. Sometimes this just moves the position and shape of the glitch.

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Defeaturing 1.2.2
  * A2plus 0.4.68
  * fasteners 0.5.33
```

### Subproject(s) affected?

CAM

### Anything else?

Here is the file: 

[AdaptivePathBug.zip](https://github.com/user-attachments/files/18916678/AdaptivePathBug.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-22:
Could you try  if the issue still exists in the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

axxel, 2025-02-23:
Yes, it also happens in Version: 1.1.0dev.40317 (Git).

But interestingly, playing around with it more, I noticed that there is some non-determinism involved, meaning switching back and forth between different Tolerance values (Accuracy vs. Performance slider) and hitting "Apply" sometimes results in different outcomes. So for the exact same value I can get a result without any of those tiny loops or with a slightly different shape or again with the exact same shape as shown in the screenshots.

dbtayl, 2025-04-04:
More broadly, Adaptive has some stability issues when small amounts of material remain and/or where paths meet from two directions. Possibly the same root cause as this issue?

I'll keep an eye out for what might cause the non-determinism as I slowly poke at the code. Nothing has jumped out at me yet.

[image]

[image]

tarman3, 2025-09-29:
I do not have experience with `Adaptive` operation

davidgilkaufman, 2025-12-12:
> I was not very hopeful that there was someone around still interested and knowledgeable about this code

There isn't, really! As far as I understand, the original authors and reviewers (if there were any?) are long gone, and knowledge has been lost. But I've done a lot of reading and experimenting with this code lately to fix bugs and improve it, and I suppose I'm becoming the new expert.

> your insights do not explain the non-determinism, right?

The details I discussed above do not, but it does make sense to me. It isn't non-deterministic per-se (same code + same input = same output), but it's certainly not stable or precise either. The adaptive algorithm is fundamentally an iterative numeric optimization algorithm, with a lot of funny patches and modifications applied to the many, many edge cases. It makes sense that small changes in the input can cause qualitative changes in the output -- a tiny difference early in the computation will cause a slightly different problem to be solved next, then a different one after that...

It won't make any difference in when/how I resolve this bug, but fyi if you'd like to support my work you can [buy me a coffee](https://buymeacoffee.com/davidgilkaufman).