FreeCAD/FreeCAD #19770: Triming not working - usual situation

mic-user, 2025-02-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

[symptom]
[image]

[error log]
10:09:14  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: SketchObject::getPoint. Invalid GeoId was supplied.
10:09:14  trim failing#group3-1: App.getDocument('trim_failing').getObject('Sketch008').trim(4,App.Vector(34.684175,-3.651028,0))
10:09:14  trim failing#group3-1: Failed to trim edge

[repro step]
1. open the atached sample
2. open the sketch name 'group3-1 / Sketch008'
3. try line trim
[trim failing.zip](https://github.com/user-attachments/files/18918010/trim.failing.zip)

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40303 (Git) Conda
Build type: Release
Branch: main
Hash: 145af5cddc9774f00fd4dfb3bd35e94efc5c70ca
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: Korean/South Korea (ko_KR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/109.201
Installed mods: 
  * Curves 0.6.54
  * lattice2 1.0.0
  * sheetmetal 0.7.5
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-22:
@AjinkyaDahale FYI
maybe related
https://github.com/FreeCAD/FreeCAD/issues?q=is%3Aissue%20state%3Aopen%20sketcher%20trimming

AjinkyaDahale, 2025-02-22:
Seems there's some conflicting/redundant constraint issue (colour of the sketch)? Does trim pass if offendong constraints are deleted?

mic-user, 2025-02-22:
@AjinkyaDahale
erm, I do not understand what you are talking about. there is none of color stuff since this is the made test-file only for reproducing the issue which usually occuring. you can check what is wrong with the attached test file.

Shkolik, 2025-02-24:
Can confirm - nothing wrong with a sketch and trim will not work. But after removing length constraint trim tool was able to successfully trim same edge.

macdroid53, 2025-04-06:
I see there are multiple issues filed for Sketcher Trim.

I just built 41155 with the following results:

Debian 12 PC (#1) any attempt to use the Trim tool crashes FC (previous source code for several days, same result)
Debian 12 PC (#2) Trim fails after second or third trim, but does not crash (yesterdays source it worked fine)
Manjaro PC (#1) works as expected any number of trims
Manjaro PC (#2) works as expected any number of trims

macdroid53, 2025-04-06:
This Trim failing after a couple trims seems to be a case specific issue.

Here it fails: 

[image]

Here it works:

[image]

AjinkyaDahale, 2025-04-06:
@macdroid53 #20150 is meant to fix this issue. If you can compile please do so from the https://github.com/AjinkyaDahale/FreeCAD/tree/sk-refactor-fix-new-trim-issues branch and confirm. In case you cannot compile and use the weekly builds, please wait, and only update after that pull request has been merged and a weekly build with those changes

Trim has taken me quite some time to rewrite into a more maintainable form, and unfortunately that rewrite had some small but still major oversights. I have spent a significant amount of time fixing them, but the fixes have not yet been merged into `main` (taking their time going through the merge process). It is possible that those fixes are still incomplete, or possibly introducing newer bugs, but so far, I haven't heard any particular comments about the fixes.

Instead, I am only hearing repeated updates that the problem still persists in a build that does not contain the fix. I have been refraining from unsubscribing from these issues so that I can hear back after the fix has been merged. However, I guess at this point I will have to start doing so.

macdroid53, 2025-04-06:
I did not know your fixes had not merged.

But, more concerning is different behavior of trim day to day and on different platforms. This implies, at least to me, that other code (not your fixes) are effecting the Trim tool.

And, why a crash on one Debian PC and not on another? Both using the same source code.

AjinkyaDahale, 2025-04-06:
@macdroid53, understood: I didn't account for the fact that my statement could suggest the fixes are merged.

Regarding the different observed behaviour, there might be some undefined behaviour stuff happening. The crash itself appears to happen because one extra trim is called when it shouldn't be, likely because the computer thinks youre moving the mouse (because of micro movements) when you aren't. It could be that different distros are differently making this guess whether or not you're moving the mouse.

closed 2025-04-10 by hyarion

mic-user, 2025-04-10:
@hyarion First, thank you for all you guys effect.
Btw, what version of release should I expect that includes the patch?

hyarion, 2025-04-10:
> First, thank you for all you guys effort.

Thanks for reporting the issue

> Btw, what version of release should I expect that includes the patch?

Dev builds (weeklies) are built on Fridays and Mondays. Next stable release with this fixed will be 1.1.0

mic-user, 2025-04-10:
@hyarion Erm, the 'weekly-builds-41213' ?? or the next?

hyarion, 2025-04-10:
@mic-user Right, timezones xD
This was merged just 2 hours ago, so next build will include it.

AjinkyaDahale, 2025-04-10:
@mic-user something after this number. 
[image]

We did have a discussion in the recent developers meeting to change the weekly numbers to something more understandable, but no promises when that will happen.

BTW, if it's already Friday in your time zone, that's like UTC+14:00. The builds happen either in European or American time zones, depending on what computer they are being built (someone else probably can pinpoint it). I guess for you it'd be best to consider it as "Tuesdays and Saturdays".

mic-user, 2025-04-10:
@AjinkyaDahale @hyarion Thanks guys.

commit 2779f0b2 (2025-04-10): [Sketcher] Add missing check in `deriveConstraintsForPieces`

Fixes #19770.