FreeCAD/FreeCAD #19776: Expression editor: Ok wrong capitalization

semhustej, 2025-02-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Expression editor has wrong capitalization of letters. 

**Feature:** change 'Ok' to 'OK' which is international standard

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40303 (Git) Conda
Build type: Release
Branch: main
Hash: 145af5cddc9774f00fd4dfb3bd35e94efc5c70ca
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Soham-47, 2025-02-23:
Hey can i work on this issue?

Syres916, 2025-02-23:
@semhustej hopefully this is the only one in the code, I tried a search for others but this was the only one I found, fix in PR #19803

closed 2025-02-23 by maxwxyz

semhustej, 2025-02-23:
> [@semhustej](https://github.com/semhustej) hopefully this is the only one in the code, I tried a search for others but this was the only one I found, fix in PR [#19803](https://github.com/FreeCAD/FreeCAD/pull/19803)

Thank you. I am aware only of this location. Other locations seem to be OK :)

PR #19803 (2025-02-23): [Gui] Expression dialog OK button capitalised and Alt+O shortcut applied
Fixes #19776 

With this fix:

[image]