FreeCAD/FreeCAD #19777: Sketch does reshape when it depends on a gear

bitcuits, 2025-02-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I am making a Sketch on a gear but when I  change the gear the sketch doesn't follow.
It seems the sketch has lost the connection to the gear.
I updated to the stable version to the dev version.
Because in the current version 1.0,0 it did not let me make a surface on a gear.

Thanks for the good work you do.
 

### Full version info

```shell
OS: Ubuntu 24.10 (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40303 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 145af5cddc9774f00fd4dfb3bd35e94efc5c70ca
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell ..7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 144/139.948
Installed mods: 
  * freecad.gears 1.3.0
```

### Subproject(s) affected?

PartDesign

### Anything else?

The drawing contains a gear and a spacer.
The spacer is created as a sketch on to of the gear.
The VarSer.outpu_n set the number of teeth on the gear.
And the ideaer is the spacer should size the diameter accordingly.
But it doesn't happen.
So the variable only affects the generated gear but not the shape which the spacer. 

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-22:
Please share an example file with the issue present. You can upload a zipped file to the original issue.
Also, add steps to reproduce the issue. Ideally, provide a macro of the issue: `Macro > Macro recording...` and repeat all steps until the issue occurs.

bitcuits, 2025-02-22:
Thanks.
I have just created the example via the macro-recording that you suggested.
[shapeongear.tar.gz](https://github.com/user-attachments/files/18923055/shapeongear.tar.gz)

bitcuits, 2025-02-25:
I tried another example with two gears.
I created two gear bodies with a shaft sketch attached to each of them.
If the number of teeth is changed on the first gear, the body will also change the body(gear1) as I would expect.
But this is not the case with the gear2, this gear body will be unchanged, so it seems that it has lost the connection with the gears.
Note.
I have also noted another thing if I have two bodies and I select something in the second body the sketch will jump to the first body and you can move it back.

Roy-043, 2025-02-25:
This is related to the Gear Workbench. Attaching to a face of a gear produced by that workbench is not TNP safe. Instead you can attach to the Placement of the gear (ObjectXY) with an offset (to account for the height of the gear). Alternatively you can attach the sketch to a subelement of the Origin of the Body the gear is in.

https://wiki.freecad.org/Topological_naming_problem

[shapeongear-roy.FCStd.zip](https://github.com/user-attachments/files/18965326/shapeongear-roy.FCStd.zip) (remove the .zip extension)

> I have also noted another thing if I have two bodies and I select something in the second body the sketch will jump to the first body and you can move it back.

Please create a separate issue for that.

bitcuits, 2025-02-25:
I think found out why the sketch fails.
It's because when the gear changes the the names of the top and bottom of the gear also change names.
This means that the sketch has the wrong reference.

closed 2025-02-26 by maxwxyz