FreeCAD/FreeCAD #19779: [BIM] Add automatic calculations of z-coordinates for overlying levels and height of elements.

kaiwas, 2025-02-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

At the moment, in the level properties we can set the height of the structures (zero by default). We can consider this to be the height of the floor.

If we create a new level, it appears in the place where the working plane is installed. That is, it repeats the position of the existing level. Also, the height of the structures (for example, a wall or a column) does not take this parameter into account.

My suggestion:
1) If we add a new level, the value of its z-coordinate is calculated automatically from the sum of the heights of the underlying levels. The connection is always maintained until the user breaks it forcibly.
It will also require some special setting for levels located below the first level (for basements). Probably a separate checkbox is needed for this.

2) The height of a wall or column added to a level (when it is active) must be equal to the "Level Height" parameter and be linked to it until the user breaks this link forcibly. This will allow you not to adjust the height of all the walls being added (even if there are a hundred of them). This value will be taken from the level height.

All this can be implemented now using expressions. But this is a rather labor-intensive process, since you have to set up such links for each newly added object (level/wall/column, etc.)

An example of such links in action on the video.

https://github.com/user-attachments/assets/b42dacc9-1a45-41f1-a15b-c33bdcc6750b

A test file is attached (remove the zip extension)

[AutoLevel.FCStd.zip](https://github.com/user-attachments/files/18922728/AutoLevel.FCStd.zip)

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40317 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b67ab979afdba808fc03196faf9c5f8830df471f
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Logical/physical DPI: 96/91.7938
Installed mods: 
  * FreeCAD_SketchArch
  * PitchedRoof
  * Road 2025.2.6
  * addFC 1.4.0
  * freecad.gears 1.3.0
  * sheetmetal 0.7.11
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-02-22:
I don't think these should be default, but it would be nice to have these options more easily accessible, than creating expressions manually. If the options were placed on the creation task panel of the wall, it could be easily set by user for walls, which will be created as new.

For reference, in Revit, user has these options:

- Set wall (beam) height as "unconnected" using a height value
- Set top constraint of the wall (which can be set to Level). User can easily select the level from drop down list
- There is also a setting for possible offset from this level
- There is also option to "attach" the wall, which extends the wall to a selected object and also cuts it according to this object's shape. This is out of the scope of this feature request.

Revit wall properties:
[image]

semhustej, 2025-02-22:
1. I don't think changing z-position of one of the levels should change z-values of the levels above by default. I think user should explicitly lock the z-positions if he/she desires them to behave this way.

kaiwas, 2025-02-22:
> 1. I don't think changing z-position of one of the levels should change z-values of the levels above by default. I think user should explicitly lock the z-positions if he/she desires them to behave this way.

perhaps all of the above should not be the default setting. maybe there are some options.
fortunately in my work everything is limited to three above-ground floors and sometimes a basement. so I can set up expressions once.
but if we are talking about rich multi-level objects - expressions become a difficult task.

for example, you can look at Archicad. there all floors and objects on them take into account heights by default.
this is very convenient.

created a floor. specified its height and in advance you know that the walls will be located at the level of this floor and have the specified height.

markkomartin, 2025-02-25:
> > 1. Non credo che cambiare la posizione z di uno dei livelli dovrebbe cambiare i valori z dei livelli soprastanti per impostazione predefinita. Penso che l'utente dovrebbe bloccare esplicitamente le posizioni z se desidera che si comportino in questo modo.
> 
> Forse tutto quanto sopra non dovrebbe essere l'impostazione predefinita... 
> 

In my heating system design software I have the automatic z-coordinate function. It is very useful because when I create a plan I can decide whether to manually write the value of the z dimension or select the automatic dimension box. 
In my opinion it could be included in the new BIM Views tool. 
In the new BIM Views tool I would like to be able to manage the heights of the individual floors too.

[image]

semhustej, 2025-02-25:
The layer manager in Archicad is neat.