FreeCAD/FreeCAD #19785: PartDesign: Pocket Cannot Taper Through All

leoheck, 2025-02-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

If the Through all is selected, there is no option for taper. This may be an implementation issue. But it could have such an option.

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40303 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 145af5cddc9774f00fd4dfb3bd35e94efc5c70ca
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell /.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/145.143
Installed mods: 
  * Assembly4 0.50.18-dev
  * sheetmetal 0.7.11
  * DynamicData 2.71.0
  * Curves 0.6.58
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * GDT
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.1.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-23:
Not sure if feasible, as depending on the angle, after a certain length the faces would overlap and create corrupted geometry.
@pierreporte is this a feature in other CADs?

pierreporte, 2025-02-23:
Don’t know if it’s a thing elsewhere but I see no reason to disallow it in FreeCAD. There are other features that can generate corrupted or impossible geometry, that generate error messages when invalid parameters are entered. The pocket feature could do it as well.

leoheck, 2025-02-23:
The through all, when there are taper angles, should extend to one side until the edges touches, the other side it can grow forever as it normally does. We can think it as a big triangle. I am not familiar with the implementation tho.