FreeCAD/FreeCAD #19795: Support STEP 242 ed1/ed2/ed3/ed4

maxwxyz, 2025-02-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Currently, FreeCAD can export and import .STEP files but has no support for the newest STEP standards defined in **ISO 10303-242** [edition 2 and edition 3](http://www.ap242.org/edition-2.html) and the upcoming [edition 4](http://www.ap242.org/edition-4.html) with **ISO/FDIS 10303-242**. Not even [edition 1 ](http://www.ap242.org/ap242ed1.html)is supported fully by FreeCAD (kinematics, PDM,...). AP242 support is critical for enabling FreeCAD to be integrated in any product life cycle workflow in the industry, which relies on this exchange format throughout the entire supply chain.

242 Edition 2/3 support besides geometric data also important pipeline date for commercial use, like 3D PMI (3D annotations and tolerances), additive manufacturing, electrical harnesses and 3D composite design.

242 Edition 4 extends support of edition 2/3 with more features for 3D PMI and assembly data, including joint and connection information which the integrated Assembly WB should be able to work with on import and export.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40122 (Git) Conda
Build type: Release
Branch: main
Hash: c1dddaeadc68cc9f05f1312dc1c6789820cfdd5d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/133.772
Installed mods: 
  * Cables-master 0.1.1
  * CfdOF 1.29.7
  * CurvedShapes 1.0.13
  * Curves 0.6.55
  * dodo 1.0.1
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.1.0
  * parts_library
  * Rocket 4.1.1
  * sheetmetal 0.7.5
  * Silk 0.1.5
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-23:
@FreeCAD/cad-advisory-group FYI

dpasukhi, 2025-02-23:
As for kinematic, there 2 ways: from XML or step.
From STP it is already possible to parse and store in session.
And I hope we will publish open-source kinematic storage and writing/reading (for now it is part of commercial extension)

wifasoi, 2025-02-24:
Unfortunally I can only export in 242 ed1 from Catia, so I can shale only PMI for testing. Will create some example file in the future

dpasukhi, 2025-02-24:
As for PMI samples, we have a lot from NIST and CAX-IF. OCCT3D joined the STP AP242 forum :)
We have a lot of samples, but if you can share more "public" files is always welcome.

dpasukhi, 2025-02-24:
How FreeCad export data from OCCT?
Using XCAF or from the Step data model? In case if XCAF it will require a lot of stuff from OCCT.
In case of step data model - we can generFe  parsers and rw interface. Processing on FreeCAD side.

WandererFan, 2025-02-24:
I know where the code is, but that's it.  Looks to be done with OCC methods.

Import module (Mod/Import/App)

AppImportPy.cpp 
Py::Object exporter(const Py::Tuple& args, const Py::Dict& kwds)
...
            Base::FileInfo file(Utf8Name.c_str());
            if (file.hasExtension({"stp", "step"})) {
                Import::WriterStep writer(file);
                writer.write(hDoc);
            }
            

void WriterStep::write(Handle(TDocStd_Document) hDoc) const 
...
    STEPCAFControl_Writer writer;

pieterhijma, 2025-03-06:
> How FreeCad export data from OCCT?
> Using XCAF or from the Step data model?
> In case if XCAF it will require a lot of stuff from OCCT. In case of step data model - we can generFe parsers and rw interface. Processing on FreeCAD side.

@dpasukhi FreeCAD has support for both methods. The import/export module @WandererFan is referring to uses XCAF but [`Mod/Part/App/TopoShape.cpp`](https://github.com/FreeCAD/FreeCAD/blob/621bebc1e3848d9ac3e3436f2544ec26844e8ef4/src/Mod/Part/App/TopoShape.cpp#L893) uses the Step data model. The import/export module that uses XCAF is the generic import/export code and is (I would assume) mostly used in FreeCAD.

If we were to move forward with this, what would you recommend to use?

pieterhijma, 2025-03-06:
> [@Kuzma30](https://github.com/Kuzma30) [@Syres916](https://github.com/Syres916) [@pieterhijma](https://github.com/pieterhijma) [@chennes](https://github.com/chennes) anyone could help here maybe?
> 
> IIRC metadata, clipping planes and PMI are currently not supported in FreeCAD (but hopefully will be in the future) - I guess we need to know how it is implemented in the file format, do you have any documentation (how metadata is stored, clipping planes, PIM)? [@FreeCAD/cad-advisory-group](https://github.com/orgs/FreeCAD/teams/cad-advisory-group) FYI

Indeed, I believe all those things are currently not supported in FreeCAD.

dpasukhi, 2025-03-08:
Thank you for the collecting information.
In that case I will try to prepare the new guide for DE components integration and extraction of attributes.
I would recommend to use XCAF in case if importing or exporting attributes for now (easier to maintain) and some custom extension to extract some extra attributes.
If you share some priorities for data types I can prepare them in first iteration.
_
During a 8.0 release I'm looking for the time to rework DE to avoid XCAF and use strict data structures. But anyway it is long term and even my POC took some time :(

pieterhijma, 2025-03-13:
Thank you very much @dpasukhi. I will bring this and the DE framework up in the coming Developers meeting this Saturday. I'm still organizing but feel free to join if you happen to have time. No problem if you can't; I can invite you to a meeting at a later stage when I'm more organized. @maxwxyz, will you be there to discuss the this issue?

- [Developers Meeting in the calendar](https://www.freecad.org/events.php)
- [Agenda Developers Meeting](https://github.com/FreeCAD/FreeCAD-developer-meetings/blob/main/Agendas/agenda-2025-03-15.md)

dpasukhi, 2026-02-07:
OpenCascade have ISO in case of some internal development and OpenCascade one of the members who participate in ISO editions. (OCCT is member of CAX-IF), we are not be able to share copy, but can share ideas how to implement in OCCT part.
The mostly step part is inside OCCT, if you have some ideas - let me know.
If you want some extension and provide your development, I can consult about what to do inside OCCT.
As for STPX I'm waiting for binary non-commercial delivery and it will be available for community.

iabnvkr, 2026-02-07:
> OpenCascade have ISO in case of some internal development and OpenCascade one of the members who participate in ISO editions. (OCCT is member of CAX-IF), we are not be able to share copy, but can [...]

can you help me understand, how it can be implemented? i'm sorry but i have never worked on something like this. can you give some advice on what should i look for? as current STEP standard in freecad is outdated and needs to be upgraded to latest. What do you think about feasiblity and its implementation?

dpasukhi, 2026-02-07:
Yes, we can have a call. Preferably on Sunday (the most part of the day) or Monday (discussable) or some other day.
_
I'm not sure the way of communication, maybe discord can works.
FreeCad is not outdated. It mostly supports big part of standard but not use :)

iabnvkr, 2026-02-07:
> Yes, we can have a call. Preferably on Sunday (the most part of the day) or Monday (discussable) or some other day. _ I'm not sure the way of communication, maybe discord can works. FreeCad is not outdated. It mostly supports big part of standard but not use :)

my discord username: `timon_pimbaa`

we can discuss whenever you are free.

dpasukhi, 2026-02-07:
Please find dpasukhi, your account close

maxwxyz, 2026-02-08:
I've updated the issue with the links shared today:
- https://www.steptools.com/stds/step
- https://www.nist.gov/services-resources/software/step-file-analyzer-and-viewer
- https://www.canva.com/design/DAHAvlvBxps/WNhjD9LNnRFedNc1HGPrTw/edit?utm_content=DAHAvlvBxps&utm_campaign=designshare&utm_medium=link2&utm_source=sharebutton
- https://github.com/Open-Cascade-SAS/OCCT
- https://github.com/Open-Cascade-SAS/Inspector

maxwxyz, 2026-03-19:
This one: https://github.com/FreeCAD/FreeCAD/issues/19795#issuecomment-3873524144