FreeCAD/FreeCAD #19798: FEM: Show results slider ignores max value

maxwxyz, 2025-02-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In the latest weekly the max slider value is ignored in the show results task dialog.
This leads to a tiny slider which I am not able to grab and slide.
Maybe a regression from https://github.com/FreeCAD/FreeCAD/pull/18496 @mac-the-bike ?

[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40317 (Git) Conda
Build type: Release
Branch: main
Hash: b67ab979afdba808fc03196faf9c5f8830df471f
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/108.888
Installed mods: 
  * Cables-master 0.1.1
  * CfdOF 1.29.8
  * CurvedShapes 1.0.13
  * Curves 0.6.58
  * dodo 1.0.1
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.1.1
  * parts_library
  * Rocket 4.1.1
  * sheetmetal 0.7.11
  * Silk 0.1.5
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-23:
@FEA-eng FYI

mac-the-bike, 2025-02-28:
**FEM: Show results slider ignores max value #19798**

I made some changes to "ResultShow.ui" 2 weeks ago, these changes were purely cosmetic, see #18496

There are three things that I notice in your picture of the "Show Result" panel:
1.  you use are using decimal commas, I have no problem with that
2.  "Show Result" panel:
    Your panel doesn't look like mine, see #18496. I have rebuilt FreeCAD today, using the latest FreeCAD source, and when I run FreeCAD, I see exactly what is seen in this picture. I am running with "Classic" and you with "Dark".
3.  your environment is: Locale: German/Germany (de_DE) and mine, as you can see, is en_GB. In my Edit-> Preferences I have: Language - English, Number Format - Operating System, decimal separator is not checked. I don't think that any of this affects the slider width.

The only suggestion I can make is: if you can get your "Show Result" panel to look like mine, then, perhaps, the slider will have a sensible width. I have no reason to think that this will work. 
This problem is outside my sphere of knowledge. If there is anybody who reads this and has some better ideas about what is wrong, then please add to this post.

Details of my build:
OS: Ubuntu 22.04.5 LTS (GNOME-Flashback:GNOME/gnome-flashback-metacity/xcb)
Architecture: x86_64
Version: 1.1.0dev.40439 (Git)
Build type: Unknown
Branch: main
Hash: fcbbe9535fd92cbc3fd855e595eb5e02195c8ddd
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/91.7938

maxwxyz, 2025-02-28:
@mac-the-bike it is not a stylesheet issue. I have switched to classic and this is how it looks:
10 as slider max and currently the slider is at 9 - so it should be 1/10th of the width and way more to the right.

[image]

It appears that the slider is not adjusted based on the slider max value, as it looks exactly the same when I set the slider max to 100,000.00:

[image]

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40421 (Git) Conda
Build type: Release
Branch: main
Hash: f9bb6508ffb420502d7b91592d87ea4f95c399d4
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/108.888
Installed mods: 
  * Cables-master 0.1.1
  * CfdOF 1.29.8
  * CurvedShapes 1.0.13
  * Curves 0.6.58
  * dodo 1.0.1
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.1.2
  * parts_library
  * Rocket 4.1.1
  * sheetmetal 0.7.11
  * Silk 0.1.5

```

mac-the-bike, 2025-03-06:
The part of the code that deals with the size of the slider was not modified by me. 
The thing I did to the ResultShow.ui interface was to add the panel for animation control.

luzpaz, 2025-04-24:
@marioalexis84 can you diagnose when you get a chance ?

maxwxyz, 2025-11-21:
@ipatch the FEM Example file works. You need to run the simulation to se the cxx_results, then double click on it in the tree view.

luzpaz, 2026-01-11:
> Also, I have had a look at the slider problem with no resolution.

Per @mac-the-bike in https://github.com/FreeCAD/FreeCAD/pull/25642#issuecomment-3670132514