FreeCAD/FreeCAD #19799: FEM: Animation is displayed in negative displacement direction

maxwxyz, 2025-02-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The FEM animation of results is showing a loop from maximum displacement to the negative of the maximum displacement, which wiggles in the wrong direction. It should only loop between 0 and max. displacement.

As you see below, the animated result is displayed as an upwards bend but the force is only applied in downwards direction which leads to a false representation:

[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40317 (Git) Conda
Build type: Release
Branch: main
Hash: b67ab979afdba808fc03196faf9c5f8830df471f
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/108.888
Installed mods: 
  * Cables-master 0.1.1
  * CfdOF 1.29.8
  * CurvedShapes 1.0.13
  * Curves 0.6.58
  * dodo 1.0.1
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.1.1
  * parts_library
  * Rocket 4.1.1
  * sheetmetal 0.7.11
  * Silk 0.1.5
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-23:
@mac-the-bike I guess from https://github.com/FreeCAD/FreeCAD/pull/18496
@FEA-eng FYI

FEA-eng, 2025-02-23:
I'm not sure I would call it a bug, more like confusing behavior. Abaqus can also animate this way but it's not the default mode:

[image]

mac-the-bike, 2025-02-28:
**FEM: Animation is displayed in negative displacement direction #19799**

I don't accept that this is a problem. 
The program works in the way it was designed to. The animation is a peak-to-peak animation. If there is a need for 0-max-0 animation, then this is a new requirement.

FEA-eng, 2025-02-28:
@mac-the-bike Yes, Max changed it to a feature request after my comment.

closed 2026-01-07 by maxwxyz