FreeCAD/FreeCAD #19807: Transform: allow transforming multiple objects at once

eblanshey, 2025-02-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The option to transform objects is disable when multiple objects are selected. The new transform tool allows moving the axes to any position, and this lends itself well to repositioning multiple objects.

For example, I downloaded this bot online, which was imported as separate objects. It will be a lot of work to mate all the limbs properly, so I want to do some quick adjustments. I selected the whole left arm and wish to rotate it about the shoulder socket, but I'm unable to. I don't want to fuse them together because they should be able to move independently.

[image]

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40317 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b67ab979afdba808fc03196faf9c5f8830df471f
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/140.99
Installed mods: 
  * WebTools 1.0.0
  * Curves 0.6.58
  * Plot 2024.11.26
  * AirPlaneDesign 0.4.1
  * Help 1.0.3
  * lattice2 1.0.0
  * CurvedShapes 1.0.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

belinski-roman, 2025-02-24:
Use the Assembly Workbench to create an Assembly container for all the parts and a Subassembly container for the arm with all its joints, and then you can move your arm even without a tool Transform.

eblanshey, 2025-02-25:
That's a great workaround, @belinski-roman, thank you! @maxwxyz should we close this, or do we still want the ability to transform multiple objects without the assembly workbench?

maxwxyz, 2025-02-25:
Yes keep it. Workaround is as simple as throwing all objects in a part container.

Begalov, 2026-09-25:
Transform should work with Groups affecting it's content .